micropython/micropython #5325: STM32 machine.bootloader() hard fault

ajaymills, 2019-11-12:
I'm trying to use the **fwupdate.py** script (from mboot) to set up Micropython firmware updates using some flash memory on my custom board as temporary storage (a dfu.gz file). While manually invoking the fwupdate.bootloader() function from the REPL (for testing) I'm observing a hard fault in my J-Link debug session.

 In modmachine.c, function machine_bootloader I step through the lines 276-278, the last of which causes the hard fault:
```
        const char *data = mp_obj_str_get_data(args[0], &len);
        void *mboot_region = (void*)*((volatile uint32_t*)0x08000000);
        memmove(mboot_region, data, len);
```

There's something about this, I can't quite follow. mboot_region will basically be set to the _estack linker symbol at the start of flash. But then, unless I'm missing something, memmove will cause a pointer march right off the end of SRAM (since _estack is always set to the end of SRAM). 

Am I missing something ? Does this work in some other context?

jimmo, 2019-11-12:
Can you tell us more about your board? Specifically which MCU, and if you've got a custom linker script.

You're right that `_estack` is at 0x08000000, but it's actually mboot's `_estack`, not the main firmware. mboot's linker script always sets the length of RAM to 96k, so on MCUs like the F7 where RAM is much bigger than that, so it isn't the end of SRAM... unless you're using an MCU with less than 96k of RAM?

ajaymills, 2019-11-12:
> Can you tell us more about your board? Specifically which MCU, and if you've got a custom linker script.
>
> You're right that `_estack` is at 0x08000000, but it's actually mboot's `_estack`, not the [...]

Hi thanks, yeah I should have included that info. I'm using the STM32L476ZET6, which has exactly 96KB in SRAM1 and 32kB in SRAM2, so they are not contiguous. For mboot I'm using the default "stm32_generic.ld" linker script and boards/stm32l476xe.ld for the main MPY firmware.

I think, I saw the 96k RAM the stm32_generic.ld and assumed it matched up with this MCU, now I understand that was set to be smaller than the actual available RAM so that _estack would actually end up somewhere in the middle  of RAM. I believe, I can fix it by just subtracting some small amount from the end of RAM. The firmware update string only appears to be 32 bytes, so doesn't seem to need much.

Thanks!

dpgeorge, 2019-11-13:
mboot is not that efficient on RAM usage, mainly because it doesn't need to be, these parts have lots of it.  Could probably reduce the 96k down to 80k to accommodate more parts.

jonnor, 2024-09-19:
Is this still relevant with recent MicroPython releases?

dpgeorge, 2024-09-20:
A board can now customise mboot's linker scripts, so it's easy for a board to reduce RAM usage.  See 58be5a5aa328e687a3560d4a85fb069400b0b8ca

closed 2024-09-20 by dpgeorge