micropython/micropython #5326: ESP32 never reaches status()==network.STAT_NO_AP_FOUND

goatchurchprime, 2019-11-12:
This issue was originally reported as [micropython-mqtt/issues/34](https://github.com/peterhinch/micropython-mqtt/issues/34) when I discovered an inconsistent behavour between the ESP8266 and the ESP32.  

Basically, when you call `network.WLAN.connect()` for a wifi access point that is not active the status() function on the ESP8266 gives `network.STAT_NO_AP_FOUND` after a certain number of seconds when it is clear the access point is not available, while the ESP32 sticks on `network.STAT_CONNECTING` forever.   

```Python
>>> import network, time
>>> w = network.WLAN()
[junk omitted]
>>> w.active(1)
>>> w.connect("BV6000", "password")
>>> time.sleep(60)   # give it a minute
... [lots of the following message]
I (108937) wifi: STA_DISCONNECTED, reason:201
no AP found
>>> w.status(), network.STAT_CONNECTING, network.STAT_NO_AP_FOUND 
(1000, 1001, 201)    # <--- ESP32 case
(3,1,3)       # <--- ESP8266 case
```
Note that the unsolicited green error messages on the command line include "reason:201", so the condition is must be known; it's just not getting through.  

Here's the same code as above, but without the outputs so it is easier to cut-and-paste onto the command line and reproduce the issue.

```Python
import network, time
w = network.WLAN()
w.active(1)
w.connect("BV6000", "password")
print(w.status())
time.sleep(60)   # give it a minute
w.status(), network.STAT_CONNECTING, network.STAT_NO_AP_FOUND 
```

Jongy, 2019-11-12:
I didn't test it on my ESP32, but from a quick look in the code I think I see the reason.

After the `WLAN.connect` call, `wifi_sta_connect_requested` is set to `true` and it remains so until `WLAN.disconnect` is called. In the event of disconnect, `event_handler` tests this value, then attempts to reconnect again, regardless of the disconnection reason. You can also see the comment in the case of `WIFI_REASON_NO_AP_FOUND` which says: `// AP may not exist, or it may have momentarily dropped out; try to reconnect.`.

In the ESP8266 case, apparently the ESP libraries handle all timeouts and reassociations. So more robust handling has to be implemented for ESP32 as well, I guess. Timeouts, at least.

dpgeorge, 2019-12-19:
I'm not sure what the conditions are for the esp8266 to get into the `STAT_NO_AP_FOUND` state, but I wouldn't think it possible to tell the difference between "no AP found" and "AP out of range right now".  From the point of view of being as reliable as possible the wifi should continue forever trying to reconnect, which is what the esp32 logic tries to do.

See eg the commit b80bccccffea7aab68d4af9580664f77fecc45b2

alex-laskin, 2020-05-07:
It is a really strange attitude: "I have my own views on reliability, so lets just hide all events from developers leaving them with no ways to find out what is going on".
Wouldn't it be better to let them decide for themselves if they want to reconnect with the same credentials or, maybe, stop trying to connect and start SoftAP for user to reconfigure device to new valid credentials? Having different ports which behave differently is also a bad thing. I want to write wifi manager once and use it both with esp32 and esp8266. And it is really strange that esp32 port has STAT_NO_AP_FOUND state, but never use it.

dpgeorge, 2021-06-17:
This is resolved by 060066804a4543e79c0bdc60a1221bb8628982e9 and 364670ecf10b2f31e9a207bc440cfb731dcfa603

closed 2021-06-17 by dpgeorge

IhorNehrutsa, 2023-11-10:
**esp32/network_wlan: Fix network.WLAN.status().** https://github.com/micropython/micropython/pull/12932
status() can return STAT_NO_AP_FOUND/STAT_WRONG_PASSWORD when ssid/password is wrong.
This allows users to set the correct credentials.