micropython/micropython #5329: ESP8266 Port: Code For Scanning Hidden Networks Removed

PsuFan, 2019-11-13:
The code that was added to scan hidden networks was removed from future commits. Scanning does not show hidden networks, or accept optional parameters in v1.11

https://github.com/micropython/micropython/blob/800ad2cdd84b3c629d053c392b877ecedda21b89/esp8266/modnetwork.c#L186

PsuFan, 2019-11-14:
Sorry got confused with all the pulls, I thought one actually got merged. I compiled code from  https://github.com/micropython/micropython/pull/2737 and it is working fine. Why didn't this get added? :)

PsuFan, 2019-11-14:
This is what I was working on, code now becomes possible to scan and connect to hidden networks with predefined mac address:

`from network import WLAN, STA_IF, AP_IF, STAT_IDLE, STAT_CONNECTING, STAT_WRONG_PASSWORD, STAT_NO_AP_FOUND, STAT_CONNECT_FAIL, STAT_GOT_IP
from time import sleep
from ubinascii import hexlify

NETWORKS = [
	['SSID', 'pwd', ['1426cb5234b0']],
	['SSID2', 'pwd2']
]

def connectWiFi():
	station = WLAN(STA_IF)
	ap = WLAN(AP_IF)

	if not station.active():
		station.active(True)

	waitForConnection()

	if station.isconnected():
		return True

	aps = station.scan(True)
	aps.sort(key=lambda ap:ap[3], reverse=True)

	for ap in aps:
		for net in NETWORKS:
			found = False
			if ap[0].decode('UTF-8') == net[0]:
				found = True
			elif len(net) == 3:
				for mac in net[2]:
					if hexlify(ap[1]).decode('UTF-8') == mac:
						found = True
						break

			if found:
				station.connect(net[0], net[1])
				waitForConnection()

			if station.isconnected():
				if ap.active():
					ap.active(False)
				return True

	if not ap.active():
		ap.active(True)

	return False

def waitForConnection():
	while station.status() == STAT_CONNECTING:
		sleep(0.25)`

dpgeorge, 2019-11-21:
> Any chance at getting this code added?

It's not a problem to add this capability (to scan hidden SSIDs), it's just that it was not a priority and there hasn't been any request for this since the first attempt to add it a while ago.

Both esp8266 and esp32 can support this feature.  The main question is whether to make hidden SSIDs always be included in the scan results, or to have a keyword argument to `scan()` to select them, like `wlan.scan(hidden=True)`.

mcauser, 2019-11-21:
+1 for not showing them by default, with hidden bool to reveal them

mcauser, 2019-11-21:
@PsuFan fyi, use 3x backticks (`) for code blocks in Github comments

dpgeorge, 2019-11-21:
> +1 for not showing them by default, with hidden bool to reveal them

Do you have reasons/arguments for not showing them by default?  (I don't have an opinion either way.)

robert-hh, 2019-11-21:
+1 for showing them all. Since they are detected by scan, they are not really hidden. And then they should be listed. Having no option in the call keeps the firmware simpler. If application code can tell whether the hidden flag is set, it can be filtered later.

PsuFan, 2019-11-21:
> 
> 
> @PsuFan fyi, use 3x backticks (`) for code blocks in Github comments

Thanks

mcauser, 2019-12-11:
> Do you have reasons/arguments for not showing them by default? (I don't have an opinion either way.)

Not really. I changed my mind - I now agree with @robert-hh

dpgeorge, 2021-08-03:
See #7611 for scanning for hidden SSIDs.

dpgeorge, 2021-08-07:
Scanning hidden APs was added in 8616129f2e382310802950db066178648a9429a3

closed 2021-08-07 by dpgeorge