micropython/micropython #5331: How to soft reset the whole context(soft reset) of upy when process running?

fei2020, 2019-11-14:
THE OLD ISSUE(#5281) IS CLOSED AND I CAN'T RE-OPEN IT BY SELF. 
I CAN'T GET REPLY FROM FORUM.MICROPYTHON.ORG. SO I POST A NEW ONE HERE. SORRY

I integrated upy into my project(Thanks for your great work). upy works well in my project process.
In one running process, the python script file(A in short) could be reloaded when it was modified. But another python script file(B in short) that imported by A could NOT be reloaded when it was modified. Therefore I want to reset "the contex" of upy.

I have tried to use mp_deinit and then re-initialize upy as the same as I do when the process startup. But when I invoke a python script again, MP_STACK_CHECK(mp_stack_check) will fail.

Thank you.

fei2020, 2019-11-14:
Now I just set MICROPY_STACK_CHECK  be  (0)as expedient temporarily. But I think it is not correct way.

dpgeorge, 2019-11-22:
A reply to this was given on the forum, see https://forum.micropython.org/viewtopic.php?f=3&t=7227