micropython/micropython #5335: USB_VCP.readline() does not read whole line

shamlian, 2019-11-15:
On a pyboard, I am having some trouble with _USB_VCP.readline()_; it seems to return whatever is in the buffer, even if it's not a full line. I would have expected the function to return _None_ until a newline character is received, as if I were reading from a file. Is this expected? I noticed that the documentation text for _USB_VCP_'s implementation of _readline()_ is different than the text for _UART_ and _socket_; I'm hoping this isn't a sign that they are supposed to work differently.

jimmo, 2019-11-18:
readline is implemented via the stream interface, and unfortunately you're right that the behavior is different between the VCP and the regular UART.

A related discussion about the regular UART here might explain some of this https://forum.micropython.org/viewtopic.php?f=6&t=7237

In the VCP case, the VCP itself has no inherent timeout (unlike the UART which has one optionally provided in `init()`), so all operations are non-blocking.

Perhaps the documentation would be clearer if it was more along the lines of "Reads available bytes up to and including a newline. Waits up to the stream's timeout for each additional byte to be available.".  (With a note to say that not all streams implement non-blocking / timeouts).

So, expected, yes. But desirable, probably not... I guess the solution is to add a timeout to the VCP instance, like `machine.UART` (i.e. `pyb_uart_obj_t`).

shamlian, 2019-11-18:
Thanks for the explanation! I think your suggestion for rewording the docs in the short term makes sense. To me, it seems like a `readline()` from an interface like a hardware or virtual serial port should be non-blocking, and return None if there are no full lines, or the line (including newline) if a full line is in the buffer. I understand why in the file I/O case you'd want to return a line without newline if it exists (last line in the file), but it seems to me like the only reason to call `readline()` on a serial port would be do get a whole line if it exists. I can't think of a good reason why you'd ever want a fraction of a line. (If you did, you could just call `read()` and piece the line together, which is what you have to do if `readline()` is not guaranteed return something ending in a newline, right? Or am I missing something?)

dpgeorge, 2019-11-18:
> I guess the solution is to add a timeout to the VCP instance

Yes I think this is the right way forward, to provide a configurable timeout.  That would be useful for `readline()` and other reads, and then the user can select the behaviour they need.

> To me, it seems like a `readline()` from an interface like a hardware or virtual serial port should be non-blocking, and return None if there are no full lines,

This is hard to do because, if the line is very long and you've already read a lot of bytes in, then it's not possible to put the bytes back.