micropython/micropython #5336: ESP32 SOLO-1 binary wanted

emard, 2019-11-15:
HI

I'd like to try ESP32 SOLO-1 module, could someone release
binaries. Currently when I load normal ESP32, it prints this:

I (455) cpu_start: Pro cpu up.
I (455) cpu_start: Application information:
I (455) cpu_start: Compile time:     Nov 15 2019 12:47:38
I (458) cpu_start: ELF file SHA256:  0000000000000000...
I (464) cpu_start: ESP-IDF:          v3.3
E (469) cpu_start: Running on single core chip, but application is built with dual core support.
E (478) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.

jimmo, 2019-11-18:
The way to add this would be to write a board config directory in `ports/esp32/boards` that references another sdkconfig file that sets `CONFIG_FREERTOS_UNICORE`.

It would look pretty much exactly like the `GENERIC` one, except with another line in `mpconfigboard.mk` to add `SDKCONFIG += sdkconfig.unicore` (which you will also have to create).

There might be other options that need to be set for this part, but the process is the same. Unfortunately I don't have one to test with.

emard, 2019-11-18:
HI

I have from ebay a 15$ board with socket for replacing ESP32 modules
so I can plug either WROOM or SOLO1 module. If you have build environment
ready just compile and upload binary on separate git repository just for this.
I can pull and test, maybe it will work immediately..

If not, it is complicated and we must wait for developers to get SOLO1 later.

dpgeorge, 2019-11-21:
It would probably make sense to create a `GENERIC_UNICORE` board definition which can be used for any board that uses a single core esp32.

I tried making one, setting `CONFIG_FREERTOS_UNICORE=y` in it, and it compiles but crashes when run on a dual-core esp32 (which it shouldn't, it should be ok to run in unicore mode on a dual core).

(In fact, unicore was the default for a long time, and in 92149c8a7954169285b2909012dc601c6e7cb0aa it was changed to have dual core the default.)

emard, 2019-11-21:
Good information!
I have tried older stable builds from download site
esp32-idf3-20190529-v1.11.bin doesn't work on SOLO-1 (constantly reboots).

latest that works on SOLO-1 is esp32-idf3-20190125-v1.10.bin

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4732
load:0x40078000,len:7496
load:0x40080400,len:5512
entry 0x4008114c
I (400) cpu_start: Pro cpu up.
I (400) cpu_start: Single core mode
I (401) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFC0E38 len 0001F1C8 (124 KiB): DRAM
I (416) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (423) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 400915EC len 0000EA14 (58 KiB): IRAM
I (435) cpu_start: Pro cpu start user code
I (6) cpu_start: Starting scheduler on PRO CPU.
OSError: [Errno 2] ENOENT
MicroPython v1.10 on 2019-01-25; ESP32 module with ESP32
Type "help()" for more information.
>>>

emard, 2019-11-21:
I tried FTP server it works!!

I tried my ECP5 project because of viper it doesn't work
but if I rewrite back to plain python it would work (very slow :)
https://github.com/emard/esp32ecp5

so yeah I'd like to see fresh new binary compiled somehow :)

UF4OVER, 2024-10-14:
###  In micropython\ports\esp32\boards folder,,has board.md, it is written: 
The following files are firmware that should work on most ESP32-based boards
with 4MiB of flash, including WROOM WROVER, SOLO, PICO, and MINI modules.

If your board is based on a WROVER module, or otherwise has SPIRAM (also known
as PSRAM), then use the "spiram" variant.

The "d2wd" variant is for ESP32-D2WD chips (with 2MiB flash), and "unicore" is
for single-core ESP32 chips (e.g. the "SOLO" modules). The "ota" variant sets
up the partition table to allow for Over-the-Air updates.

you can try "unicore" file