micropython/micropython #5337: File erased on abrupt power in ESP32

hetvishah08, 2019-11-16:
Hello,
I'm using ESP32 Devkit C with the latest Micropython version. On abrupt power, all the files in the filesystem like - main.py, boot.py and other files get erased on the ESP32.
 
Opening the port through UPyCraft IDE, I wasn't able to access the device folder. On using Putty to check the files using  **os.listdir()** , I get garbage values in it.
   
```

os.listdir()

['\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x0 0\x00\x00','\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x0 0\x00\x00']                                                                                   

```
Brown-out detector might be a solution to protect the ESP32 from power changes on hardware side but I would like to know the software solutions of saving the ESP32 code memory from getting erased.

peterhinch, 2019-11-16:
This should not occur unless you're writing to the filesystem at the time the power goes down.

robert-hh, 2019-11-16:
Since recently, the ESP32 file system also supports littlefs2, which claim to be more robust. You have to use the daily build for that. A short note on using it is here: https://github.com/micropython/micropython/commit/4be316fb072ab7c2f28da815ead970ebf705c328

hetvishah08, 2019-11-18:
> This should not occur unless you're writing to the filesystem at the time the power goes down.

  @peterhinch  No I have uploaded the code successfully while the circuit was receiving power.After that,the code was working fine till some time. As soon as there was power change,all the files got erased.

hetvishah08, 2019-11-18:
> Since recently, the ESP32 file system also supports littlefs2, which claim to be more robust. You have to use the daily build for that. A short note on using it is here: [4be316f](https://github.com/micropython/micropython/commit/4be316fb072ab7c2f28da815ead970ebf705c328)

@robert-hh  I'll try it out.Thank you.

nevercast, 2019-11-18:
Interesting that the file names and extentions have been zeroed but the file counts and name lengths seem to be intact.

hetvishah08, 2019-11-19:
> Since recently, the ESP32 file system also supports littlefs2, which claim to be more robust. You have to use the daily build for that. A short note on using it is here: [4be316f](https://github.com/micropython/micropython/commit/4be316fb072ab7c2f28da815ead970ebf705c328)

@robert-hh  It worked.Thank you

closed 2019-11-21 by dpgeorge