micropython/micropython #5340: ESP32 software SPI LSB first wanted

emard, 2019-11-17:
Currently software SPI channel no. -1 doesn't allow LSB first operation.
What only works is MSB first. For performance reasons, I need both
MSB first and  LSB first in software SPI so I'd like here to post a feature request :-)

nevercast, 2019-11-18:
Am I mistaken? I don't think there is any software SPI on ESP32. Its all hardware SPI. Can you show an example of the code you're using and the output you get when running it?

emard, 2019-11-19:
hm you should lnow better :)
this is how I initialize what I think is software SPI, it has -1 channel number
and it accepts only firstbit=SPI.MSB and I'd like to set SPI.LSB sometimes:

self.swspi=SPI(-1, baudrate=self.spi_freq, polarity=1, phase=1,
bits=8, firstbit=SPI.MSB, sck=Pin(self.gpio_tck),
mosi=Pin(self.gpio_tdi), miso=Pin(self.gpio_tdo))

If I set firstbit=SPI.LSB it returns error:
ValueError: firstbit must be MSB

nevercast, 2019-11-19:
Software SPI support was an external module all this time and I'd never noticed. Interesting! I was incorrect. extmod/machine_spi.c should be your soft SPI

emard, 2019-11-19:
Great, if it can be fixed to accept LSB and MSB I'll be happy :)

dpgeorge, 2024-07-12:
LSB for SoftSPI was implemented in ee1036023ef199636d96e70c6c7ed587ccaab92e

closed 2024-07-12 by dpgeorge