micropython/micropython #5344: [ESP32] utime.sleep_ms does not free GIL on small intervals

nevercast, 2019-11-20:
When using multithreading on the ESP32 port, and using `sleep_ms(1)` to yield to another thread that may be waiting on the GIL, the sleep_ms implementation does not free the GIL when milliseconds is approximately less than 10ms. This is because there is an early break from the sleep loop that doesn't call `MICROPY_EVENT_POLL_HOOK`

https://github.com/micropython/micropython/blob/master/ports/esp32/mphalport.c#L120-L138

By changing the inner loop from 
```
    for (;;) {
        uint64_t t1 = esp_timer_get_time();
        dt = t1 - t0;
        if (dt + portTICK_PERIOD_MS * 1000 >= us) {
            // doing a vTaskDelay would take us beyond requested delay time
            break;
        }
        MICROPY_EVENT_POLL_HOOK
        ulTaskNotifyTake(pdFALSE, 1);
    }
```

to 

```
    for (;;) {
        MICROPY_EVENT_POLL_HOOK
        uint64_t t1 = esp_timer_get_time();
        dt = t1 - t0;
        if (dt + portTICK_PERIOD_MS * 1000 >= us) {
            // doing a vTaskDelay would take us beyond requested delay time
            break;
        }
        ulTaskNotifyTake(pdFALSE, 1);
    }
```

The GIL will always be freed from this thread, even if the sleep_ms duration is very small, allowing another thread to access python objects and perform work that the main thread may be waiting on.

dpgeorge, 2019-11-21:
> When using multithreading on the ESP32 port, and using `sleep_ms(1)` to yield to another thread that may be waiting on the GIL, the sleep_ms implementation does not free the GIL when milliseconds is approximately less than 10ms.

It's arguable as to whether `sleep_ms(1)` should yield, because if you make such a call it's presumably because you want a short 1ms delay.  Yielding would probably make this delay a lot longer than 1ms.

There are possibly better ways to coordinate multiple threads.  Eg using `_thread.allocate_lock()`, using semaphores, etc.  Note that the GIL is released by the VM regularly and shouldn't be something that the user is concerned about.

And note that `machine.idle()` does yield to the RTOS, but doesn't release the GIL (and maybe it should...).

nevercast, 2019-11-21:
> Note that the GIL is released by the VM regularly and shouldn't be something that the user is concerned about. 

Then this functionality is not working, because my threads lock each other indefinitely unless:
* I sleep for more than 10 milliseconds such that the hal_sleep_ms implementation does not early exit and calls the POLL_HOOK
* I call `print()` with a long string such that it calls POLL_HOOK

It was my understanding that `sleep_us` is for accurate timing and `sleep_ms` is guaranteed to sleep for *at least* as long as specified, but cannot assure accuracy. Though the documentation doesn't make that clear, that truely is the nature of the two functions. (Sleeping for 100ms will not be accurately 100ms for example).

The documentation currently writes:

> Delay for given number of milliseconds, should be positive or 0.

Which doesn't give much information on the behaviour of zero or what happens while sleeping. 

> And note that machine.idle() does yield to the RTOS, but doesn't release the GIL (and maybe it should...).

Perhaps `machine.idle()` is a good candidate for thread yielding, however I'm still of the opinion that sleep_ms should surrender the GIL to other threads also. Currently, none of the following wait examples will work, they all lock the GIL 

```
class Worker:
  def thread_worker(self):
    do_work()
    self.work_complete = True 

  def wait_with_sleep(self):
    while not self.work_complete:
      time.sleep_ms(5)
  
  def wait_with_no_sleep(self):
    while not self.work_complete:
      time.sleep_ms(0)
  
  def wait_idle(self):
    while not self.work_complete:
      machine.idle()
```

nevercast, 2019-11-21:
Since implementing the PRs I have made on my own local build my threading has been working as I expect, however one extra thing I should mention is that if the thread_worker wishes to handle interrupts, and do something useful with the interrupt handler, it might use `micropython.schedule` to do that work. It will be scheduled and it will run, but it will not have the GIL and be unable to access any state. This is an issue I faced where an IRQ from some radio hardware would indicate that data was available, `micropython.schedule` was called to read the data and push it to the queue. The scheduled event would lock up on the `self.queue.push`

dpgeorge, 2019-11-21:
> Currently, none of the following wait examples will work, they all lock the GIL

Could you please give a full working example to replicate the issue?

jimmo, 2019-11-26:
Summary: @nevercast `machine.idle()` probably does what you want (yield), despite holding the GIL while it does so. But it would be much better to use a lock to signal completion rather than polling a variable (see example at the end).

But some background relevant to this issue for a situation where you have two threads -- one working, the other waiting for a flag to be set -- and the waiting thread causes a slowdown of the working thread. (Repro below)

As @dpgeorge says -- " the GIL is released by the VM regularly" -- the VM itself will bounce the GIL every 32 operations. So if you have two threads:
 - One is computationally bound. It runs for 32 ops, then bounces the GIL.
 - The other is sleeping, waiting for the computation. Either using: sleep(0), sleep(1 <= n <= 10), sleep(>=11)

Considering the three sleep() cases as implemented currently:
 - sleep(0) is a no-op. The loop just does 32 vm ops of busy work, and then the VM yields to the other thread. Overall the result is that you've halved the throughput of the working thread.
 - sleep(1..10) is a busy loop. Now the result is that every N ops (whatever "while True: sleep(1)" is in opcodes), you do a couple of ms of busy looping. Which just has the effect of stealing more CPU time for the waiting thread. i.e. the working thread still does its 32 ops, but less often -- think of it as one of your ops (call function) taking longer. The throughput of the working thread decreases significantly. (I measure about a 20x slowdown)
 - sleep(>=11) this makes `mp_hal_delay_ms` invoke `ulTaskNotifyTake`, which at least means that the FreeRTOS scheduler will yield, and give more time to the working task. But the overall throughput of the working task is still approximately halved (my testing gets very close to half).

So the change to make `mp_hal_delay_ms` proposed in this issue to unconditionally bounce the GIL improves the second case, essentially making it the same as the first and third cases. (i.e. both threads get equal CPU time).

But interestingly, if you change `time.sleep_ms()` to `machine.idle()` as suggested, the slowdown  disappears (i.e. the computation runs as fast as it would without the second thread). This is a bit surprising, because `machine.idle()` yields the RTOS thread while still holding the GIL (as @dpgeorge pointed out above). However, it still helps because the very next time it releases the GIL, it then forfeits the rest of its time slice, so the working task gets full 10ms time slices, while the waiter thread gets very short time slices. This surprised me, but the instrumentation I added to investigate this suggests that releasing a mutex on ESP32/FreeRTOS doesn't yield the thread -- i.e. I see each thread get several goes at the GIL before the other thread runs, alternating on a 10ms tick interval (i.e. `CONFIG_FREERTOS_HZ`). Any FreeRTOS/ESP32 experts know more about this?

This suggests that releasing the GIL around the yield would offer a small improvement, and indeed it does. In the example below, `idle` is unmodified, `noop` is an empty function (so the bytecode is the same), and `idle_yield` is the same as `idle` but releases the GIL. I agree it probably makes sense to make this change to `idle`.

```python
import _thread
from time import sleep_ms, ticks_ms, ticks_diff
from machine import idle, idle_yield, noop

done = False

def wait():
    print('start wait thread')
    t = ticks_ms()
    while not done:
        # idle()   # Finishes in 611ms
        # noop()  # Finishes in 1197ms
        idle_yield()  # Finishes in 594ms
        # sleep_ms(1)  # Finishes in 3120 ms  (1185ms with suggested change)
        # sleep_ms(10)  # Finishes in 20785 ms  (1227ms with suggested change)
        # sleep_ms(11)  # Finishes in 1194 ms
    print('wait', ticks_diff(ticks_ms(), t))

def work():
    global done
    _thread.start_new_thread(wait, ())
    sleep_ms(20)
    print('start work thread')
    t = ticks_ms()
    for i in range(512):
        for j in range(512):
            pass
    done = True
    print('work', ticks_diff(ticks_ms(), t))

work()
sleep_ms(500)
```

Here's how this would be better implemented with a lock instead (with the caveat that there's currently no timeout available for lock.acquire() as already discussed in #3332 )

```python
import _thread
from time import sleep_ms, ticks_ms, ticks_diff

signal = _thread.allocate_lock()

def wait():
    print('start wait thread')
    t = ticks_ms()
    signal.acquire()
    signal.release()
    print('wait', ticks_diff(ticks_ms(), t))

def work():
    signal.acquire()
    _thread.start_new_thread(wait, ())
    sleep_ms(20)
    print('start work thread')
    t = ticks_ms()
    for i in range(512):
        for j in range(512):
            pass
    signal.release()
    print('work', ticks_diff(ticks_ms(), t))

work()
sleep_ms(500)
```

dpgeorge, 2021-07-20:
See #7554 for a fix (for both `machine.idle()` and `time.sleep_ms()`).

dpgeorge, 2021-07-21:
Fixed by db6d60b07967f6d75d74349b021b262cb8fee628 and 2cfbe5bc0fe2c8ed24c6f35abbc82ef8415d17e4 and 6214fa3f9ecdfb146aaec09afc9d586e76257d5a

closed 2021-07-21 by dpgeorge