micropython/micropython #5345: utime.sleep_ms does nothing useful when called with 0 milliseconds.

nevercast, 2019-11-20:
While debugging #5344 I encounted a situation where `utime.sleep_ms(0)` did not free the GIL to another thread. This is because `utime.sleep_ms(0)` is a useless invocation that does nothing.

https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/extmod/utime_mphal.c#L51

```
STATIC mp_obj_t time_sleep_ms(mp_obj_t arg) {
    mp_int_t ms = mp_obj_get_int(arg);
    if (ms > 0) {
        mp_hal_delay_ms(ms);
    }
    return mp_const_none;
}
```

I propose that at the very least this function calls `MICROPY_EVENT_POLL_HOOK` when `ms` is zero, and at best, always calls `mp_hal_delay_ms`. I don't believe zero should be a special case.

dpgeorge, 2021-07-21:
> I don't believe zero should be a special case.

I agree.  It should have "continuous" behaviour for `sleep_ms(x)` where x tends to 0.

dpgeorge, 2021-07-21:
Fixed by db6d60b07967f6d75d74349b021b262cb8fee628

closed 2021-07-21 by dpgeorge