micropython/micropython #5350: How could I wake up my Micropython?

459548764, 2019-11-20:
I want to set Pin.A0 as the interrupt pin. When set wake = machine.DEEPSLEPP that it will occur an error that 

  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'DEEPSLEEP'

What should I do to solve this error?

459548764, 2019-11-20:
>>> p0 = Pin("A0",Pin.IN)
>>> p0.irq(trigger=Pin.IRQ_RISING,wake=machine.DEEPSLEEP)

dpgeorge, 2019-11-21:
I assume you are on an stm32 board.  In that case an IRQ on PA0 should automatically wake the MCU from lightsleep and deepsleep, by just doing `p0.irq(trigger=Pin.IRQ_RISING)`

459548764, 2019-11-26:
I  use the code
stm.mem32[stm.PWR+stm.PWR_CSR] |= 1 << 8
It works
However I met another situation that I want the RTC wake up my stm32. I set 
rtc = pyb.RTC()
rtc.wakeup(60000)
BUT it can not wakeup. Meanwhile the button interrupt failes as well. How could I do solve this problem?

dpgeorge, 2019-11-27:
These types of questions are best asked at https://forum.micropython.org.  And it's also good to specify the version of MicroPython that you use, which board, and also provide self contained code so others can reproduce any issues you see.