micropython/micropython #5351: Unable to build mpy-cross using gcc 9.2.1

lsangild, 2019-11-21:
After a clean clone of the repo mpy-cross fails (tried multiple commits, so expects the issue is the compiler.
A make log with V=1 is attached.
[make_log.log](https://github.com/micropython/micropython/files/3874731/make_log.log)

dpgeorge, 2019-11-22:
Indeed it looks like the toolchain you're using has some problems.  (Build works ok for me on Arch Linux with gcc 9.2.0.)

As a workaround, try building with `make STRIP=echo`

lsangild, 2019-11-22:
Your workaround works, but this might be an issue to consider solving in the makefiles as more users will undoubtedly run into this issue over the following months.

jimmo, 2019-11-24:
See https://bugzilla.redhat.com/show_bug.cgi?id=1774483 -- looks like a (short-lived) binutils bug.

closed 2019-11-24 by jimmo