micropython/micropython #5352: ESP32: Not detected as a drive/usb/device

Andrew-ux, 2019-11-21:
I have been trying to flash micro python onto an ESP32. I am able to go through all the steps, and complete the flash process. 
The dilemma is that, when plugged in, there is not a device that appears in file explorer. I know that it is supposed to have one with boot.oy, and main.py, but there is no device shown. In device manager, it appears as a port, the USB to UART bridge.
I have tried using uPyCraft as an alternative, but it needs a directory to go to to actually use, which is non-existent.
I know that I am using a data cable, because I was able to properly flash the device. It also plays the USB plug in sound when plugged in, and unplugged.

robert-hh, 2019-11-21:
The ESP32 does not show up with a file system in the file explorer, since it only has a UART connection to the host. The various PyBoard connect with a file system. You may use rshell or pyboard.py for file transfer.

dpgeorge, 2019-11-22:
This type of question is best for https://forum.micropython.org

closed 2019-11-22 by dpgeorge