micropython/micropython #5355: Http request fails with ENOMEM

DonJayamanne, 2019-11-24:
Got an ESP32 (M5Stack), and making a simple Https request fails.
Tried soft reset and reboot.
Copied source from `urequests` and ran it manually with `gc.collect` at everypoint.
Still fell over at the same place.

Falls over when executing `s = ussl.wrap_socket(s, server_hostname=host)`.
Bombs with `Error 12, ENOMEM`

jimmo, 2019-11-24:
Which firmware version are you using? How many concurrent HTTP requests are you doing and/or does it always fail on the first request or only on subsequent ones?

This is quite possibly related to #5303 , in which case you might have more luck with an IDF3 build, or a newer IDF4 build.

FYi, despite the ENOMEM error, the memory used by wrap_socket doesn't come from the MicroPython heap, which is why gc.collect() does not help.

enesbcs, 2021-05-14:
Seems similar than https://github.com/micropython/micropython/issues/7038

jimmo, 2022-07-21:
This issue is now being tracked in https://github.com/micropython/micropython/issues/8940.