micropython/micropython #5356: lwip git host down, perhaps make a mirror?

andrewleech, 2019-11-25:
As I write this, https://git.savannah.gnu.org/ is down as it surprisingly often is.
The `lwip` project is hosted here, so is needed to clone the submodule.

The host being down means CI builds of micropython fail fairly regularly

@dpgeorge would you consider mirroring the lwip project here in your github group to increase the stability of access?

dpgeorge, 2019-11-25:
See also #4429 for past issues with the lwip URL.

dpgeorge, 2019-11-25:
> would you consider mirroring the lwip project here in your github group to increase the stability of access?

Yes I guess this makes sense.  It is quite limiting not being able to develop when a site goes down.

All other submodules link to a GitHub URL.  Of course that means we have a big dependence on GitHub being up, but it has a history of being pretty stable.  And we have to host the code somewhere and rely on something, so GitHub seems a sensible choice.

Probably someone else already cloned lwip to GitHub and we can just link to that?

andrewleech, 2019-11-25:
I had a quick look and didn't find any github lwip's that looked particularly "official", most notably the stmduino one is modified, the other one that came up hadn't brought across the tags.

andrewleech, 2019-11-25:
Actually, here's a couple of candidates:
* https://github.com/ARMmbed/lwip
* https://github.com/particle-iot/lwip

sniperini, 2021-05-11:
Seems like there is an official github mirror now for LWIP repo: https://github.com/lwip-tcpip/lwip

Maintainers seems to be same people as in savannah repo.

Could we perhaps move switch the link for lwip submodule to use github repo now?

dpgeorge, 2021-05-11:
> Could we perhaps switch the link for lwip submodule to use github repo now?

Yes, sounds good!

sniperini, 2021-05-11:
Great. Thank you for prompt reply. I too hope it's more reliable as well!

dpgeorge, 2021-05-13:
Done in 43e7e5f00a601b36b5b39f1fef036d6c25d5562c

closed 2021-05-13 by dpgeorge

Gadgetoid, 2021-06-05:
Came here to suggest a switch to a git mirror, glad to see it's in hand! I'm building against the latest released MicroPython (v1.15) and, predictably:

```
fatal: unable to access 'https://git.savannah.gnu.org/r/lwip.git/': Failed to connect to git.savannah.gnu.org port 443: Connection timed out
fatal: clone of 'https://git.savannah.gnu.org/r/lwip.git' into submodule path '/home/runner/work/pimoroni-pico/pimoroni-pico/micropython/lib/lwip' failed
```

I guess I could hotfix the .gitmodules with some `sed` magic, but is a v1.16 release imminent?

dpgeorge, 2021-06-06:
> is a v1.16 release imminent?

Yes it should be out within the coming week.