micropython/micropython #5360: Pyboard 1.x DAC buffer corruption

peterhinch, 2019-11-26:
I'm not sure if this is a bug or a case of _caveat implementator_, but it is potentially confusing. The following can fail after a period with a corrupted buffer:
```python
def x5_test_signal(amplitude, freq):
    dac = DAC(1, bits=12)  # X5
    buf = array('H', 2048 + int(amplitude * math.sin(2 * math.pi * i / 128)) for i in range(128))
    tim = Timer(2, freq=freq * len(buf))
    dac.write_timed(buf, tim, mode=DAC.CIRCULAR)
```
A scope shows that, after a period, the output changes from a sinewave to garbage. The problem seems to be that after the function completes, all references to **buf** are lost so its RAM is eventually re-used. Returning **buf** and assigning it to a variable in the calling routine fixes the problem.

The following demonstrates this, producing incorrect results after a short period of running (link X4 and X5 on a Pyboard 1.x):
```python
import time
import micropython
import gc
micropython.alloc_emergency_exception_buf(100)

class Goertzel:
    def __init__(self, adc, nsamples, freq, sampling_freq, verbose=False):
        if verbose:
            print('Freq {}Hz +- {}Hz'.format(freq, sampling_freq / (2 * nsamples)))
        self.calc_ref = self.calc
        self.sampling_freq = sampling_freq
        self.buf = array('H', (0 for _ in range(nsamples)))
        self.idx = 0
        self.nsamples = nsamples
        self.adc = adc
        k = round(nsamples * freq / sampling_freq)
        self.scaling_factor = nsamples / 2.0
        omega = 2.0 * math.pi * k / nsamples
        self.coeff = 2.0 * math.cos(omega)
        self.fac = -math.cos(omega) + 1j * math.sin(omega)
        self.acquire()

    def acquire(self):  # nonblocking acquisition
        self.idx = 0
        self.tim = Timer(6, freq=self.sampling_freq, callback=self.tcb)

    def tcb(self, _):  # Timer callback
        buf = self.buf
        buf[self.idx] = self.adc.read()
        self.idx += 1
        if self.idx >= self.nsamples:
            self.tim.deinit()

    def calc(self):
        coeff = self.coeff
        buf = self.buf
        q0 = q1 = q2 = 0
        for s in buf:  # Loop over 200 samples takes 3.2ms on Pyboard 1.x
            q0 = coeff * q1 - q2 + s
            q2 = q1
            q1 = q0
        self.acquire()
        return cmath.polar(q1 + q2 * self.fac)[0] / self.scaling_factor

# Create a sinewave on pin X5
def x5_test_signal(amplitude, freq):
    dac = DAC(1, bits=12)  # X5
    buf = array('H', 2048 + int(amplitude * math.sin(2 * math.pi * i / 128)) for i in range(128))
    tim = Timer(2, freq=freq * len(buf))
    dac.write_timed(buf, tim, mode=DAC.CIRCULAR)

def test(amplitude=2047):
    freq = 1000
    x5_test_signal(amplitude, freq)
    adc = ADC(Pin.board.X4)
    g = Goertzel(adc, nsamples=100, freq=freq, sampling_freq=10000, verbose=True)
    while True:
        time.sleep(0.5)
        print(g.calc())
```

dpgeorge, 2019-11-27:
> The problem seems to be that after the function completes, all references to **buf** are lost so its RAM is eventually re-used

I didn't try out the test code but I'm pretty sure this is exactly what is happening: the GC is reclaiming **buf** while it's still being used by the DAC.

> I'm not sure if this is a bug or a case of _caveat implementator_

IMO this should be accepted as the latter, that the code must maintain a reference to any buffer that is used "in the background" by the hardware.

This will be a general problem when more peripherals support async/background execution, eg SPI background writes.  In all such cases the peripheral would be required to hold a reference to the buffer/object that it uses, which adds overhead to the implementation of that peripheral.  And it can be easily done in Python by just keeping a reference to the buffer/object, which in some cases is done anyway (eg because the buffer will be reused later).

I agree it's a bit confusing when things go wrong with GC reclaiming still-used objects, but this could be addressed in the docs.

peterhinch, 2019-11-28:
Fair enough.

closed 2019-11-28 by peterhinch