micropython/micropython #5368: WIPY1: Executing mpy files fails

robert-hh, 2019-11-29:
Cross compiled files cannot be executed any more on a WiPy1.  Error message:
```
>>> import hello
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: decimal numbers not supported
```
Firmware version:
MicroPython v1.11-611-g7f24c2977-dirty on 2019-11-29; WiPy with CC3200

Sample test code, definitely w/o a decimal number:

```
def hello():
    print("Hello WiPy1")

hello()
```

dpgeorge, 2019-12-02:
I tested this and it works ok for me:
```
MicroPython v1.11-611-g7f24c2977-dirty on 2019-12-02; WiPy with CC3200
Type "help()" for more information.
>>> import sys
>>> sys.implementation
(name='micropython', version=(1, 11, 0), mpy=517)
>>> import os
>>> os.listdir()
['main.py', 'sys', 'lib', 'cert', 'boot.py', 'hello.mpy']
>>> import hello
Hello WiPy1
>>> 
```

robert-hh, 2019-12-02:
Same situation. Firmware built today. mpy-cross version v5. 
```
MicroPython v1.11-611-g7f24c2977-dirty on 2019-12-02; WiPy with CC3200
>>> sys.implementation
(name='micropython', version=(1, 11, 0), mpy=517)

./mpy-cross --version
MicroPython v1.11-611-g7f24c2977-dirty on 2019-12-02; mpy-cross emitting mpy v5

```

robert-hh, 2019-12-02:
Same for a firmware image loaded from the web site. Must be related to mpy-cross.

robert-hh, 2019-12-02:
It was a PEBKAC case. I had an old version of mpy-cross dangling around.

closed 2019-12-02 by robert-hh