micropython/micropython #5369: PPP tunel for WiFi

emard, 2019-11-29:
HI

say ESP32 connects as client to home WiFi access point.

can ESP32 provide PPP internet access for linux over RX/TX serial?
(or over something similar like SPI channel)

Is there some example for this on micropython, please help

MrSurly, 2019-11-29:
The _intention_ of the PPP module is to provide a PPP _client_ for the ESP32.  I'm not sure if it will route packets the other way.  It uses LWIP PPP under the hood.

emard, 2019-12-01:
Some PPP basic functionality will be really welcome, like SLIP

I don't need any of fancy PPP server features as PAP/CHAP
authentication. Automatic IP address for both endpoints would
be nice but even that I can set manually

I need just elementary IP packet tunnel to WiFi router
if someone has some example please share!

emard, 2019-12-01:
Here is something mentioned about SLIP, if something as that
would appear with some example for ESP32 I'd be happy:

https://forum.micropython.org/viewtopic.php?t=788&start=20

emard, 2019-12-02:
Here is a ESP8266 project that provides SLIP tunnel
to WiFi router, something like this would be great to have
from micropython:

https://github.com/martin-ger/esp_slip_router

emard, 2019-12-02:
Just for reference, there's also some NAT working in this direction....
https://github.com/jonask1337/esp-lwip

emard, 2019-12-04:
HI!

I managed that linux PPP connects to ESP32 and can access
e.g. FTP server running on ESP32. Now my question is do I need
to setup somehow ESP32 to route packets to upstream wifi access
point in order to provide linux with PPP access to outside internet...

https://github.com/emard/esp32ppp

MrSurly, 2019-12-05:
What you're attempting is a rather unusual use case for MicroPython.  I believe what you want to do is _possible_, but implementation is non-trivial.  Additionally, even if it were to be made to work, it's not likely to be merged into the ESP32 port, since it ultimately relies on the standard ESP-IDF distribution.  That distribution has stock LWIP, which does not support NAT.

Not sure what your end goal is, but one way to implement the PPP \<--\> NAT \<--\> WiFi would be to use a Raspberry Pi Zero W, which as WiFi, several 3.3v compatible GPIO (including SPI/UART/etc), and runs a proper Linux kernel that would be relatively easy to configure for doing NAT.

emard, 2019-12-05:
I can summarize this issue in one line

I'd like if ESP32 official binary is built with

#define IP_FORWARD 1

Paul Ruiz compiled it and it does what I need.
Binary is here so you can check how it works:
https://gitlab.com/pnru/ulx3s-misc/blob/master/upython/upython_pnr5.bin

I can live without NAT. but if one feels lucky to make it,
yes it would be very useful of course.

dpgeorge, 2019-12-06:
> I'd like if ESP32 official binary is built with
> 
> #define IP_FORWARD 1

It doesn't look like the IDF allows configuration of that lwIP-specific config, so it can't be done without a custom IDF.

It's possible to built it yourself with a custom IDF and a custom esp32 board definition.