micropython/micropython #5371: reversed() does not work as expected for ucollections.OrderedDict

nevercast, 2019-12-02:
Since I expected OrderedDict to be a doubly linked list, and I needed the last element I wanted to use `reversed` to grab the last element in the OrderedDict, it didn't work as expected so I wrote some test code. 

```
from collections import OrderedDict
from random import randrange

source = list(set(randrange(2000) for _ in range(2000)))
od = OrderedDict((k, 1) for k in source)

print(list(od) == source) # True 

# CPython: True
# MicroPython: KeyError or False
last_key = next(reversed(od))
print(last_key == source[-1]) 

# CPython: True
# MicroPython: KeyError
print(list(reversed(source)) == list(reversed(od))) # True
```

Sure enough, it does not work expected. Producing either KeyErrors or just bad values.

nevercast, 2019-12-02:
Interestingly, when I changed my code to be the following, which I believe is still valid in CPython. ESP32 hard crashes. The code path for ESP32 shouldn't have changed, `import micropython` should succeed and it should work normally. 

```
from collections import OrderedDict
from random import randrange

source = list(set(randrange(2000) for _ in range(2000)))
try:
  import micropython
  od = OrderedDict([(k, 1) for k in source])
except:
  od = OrderedDict({ k:1 for k in source})
  

print(list(od) == source) # True 

# CPython: True
# MicroPython: KeyError or False
last_key = next(reversed(od))
print(last_key == source[-1]) 

# CPython: True
# MicroPython: KeyError
print(list(reversed(source)) == list(reversed(od))) # True
```

nevercast, 2019-12-02:
The primary issue seems to be that `dict` and therefore `OrderedDict` (since they are both implemented in https://github.com/micropython/micropython/blob/master/py/objdict.c) does not support `__reversed__` which is how CPython impements `reversed()` on `OrderedDict`

nevercast, 2019-12-02:
It seems that `reversed()` will try to iterate a `dict` also which will cause the same `KeyError`

nevercast, 2019-12-02:
I have an initial implementation here: https://github.com/nevercast/micropython/tree/fix/dict-reversed but I have not fully tested it so I've not made a PR. User testing seems okay though.

[image]

nevercast, 2019-12-06:
Thanks for doing an initial review of my branch, @dpgeorge. Shall I bring it in to a PR so that we continue the review there instead?

nevercast, 2019-12-06:
When considering this issue, I think https://github.com/micropython/micropython/pull/5323 should also be considered. Python 3.6 has ordered `dict`s as a side effect, Python 3.7 has ordered `dict`s by design and Python 3.8 has reversable `dict`s and `dict_view`s. Typically, MicroPython targets Python 3.6 but in this case it might make sense to consider keeping everything ordered and reversable (like Python 3.8) if it simplifies the implementation and doesn't cause any regressions that break Python 3.6 compatibility.

dpgeorge, 2019-12-09:
> Shall I bring it in to a PR so that we continue the review there instead?

Yes, if you like.

> but in this case it might make sense to consider keeping everything ordered and reversable

It has a big performance hit, making dicts simple ordered arrays.