micropython/micropython #5373: NRF51822 - is I2C working?

bgolab, 2019-12-03:
Hi,
Just flashed the MicroPython on PCA10028 compatible board.
I tested this board using C and I2C works fine.
Also tested the I2C module on another board (STM32) with success

When I try the I2C it fails:
>>> from machine import I2C, Pin
>>> i2c=I2C(-1, scl=Pin(1), sda=Pin(0))
>>> i2c.scan()
[]

>>> from machine import I2C, Pin
>>> i2c=I2C(0, scl=Pin(1), sda=Pin(0))
>>> i2c.scan()
[9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 31, 33, 35, 37, 39, 41, 43, 45, 47, 49, 51, 53, 55, 57, 59, 60, 62, 64, 66, 68, 70, 72, 74, 76, 78, 80, 81, 82, 83, 84, 85, 86, 87, 89, 91, 93, 95, 97, 99, 101, 103, 105, 107, 109, 111, 113, 115, 117, 119]
>>>

So I wonder if someone tested the I2C on NRF51 board.
When use

bgolab, 2019-12-03:
Just flashed pyboard with 1.11 and I2C works with the same I2C module:
>>> from machine import I2C, Pin
>>> i2c=I2C(1)
>>> i2c.scan()
[76]

bgolab, 2019-12-03:
Problem solved!
- I looks like the scan shows too many addresses - whereas the pyboards does it right - see the previous post.
- only option '0' works fine; the '-1' shows nothing if it comes to scanning

>>> MicroPython v1.11-611-g7f24c2977-dirty on 2019-12-02; PCA10028 with NRF51822
Type "help()" for more information.
>>> from machine import I2C, Pin
>>> i2c=I2C(0, scl=4, sda=3)
>>> i2c.scan()
[9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 31, 33, 35, 37, 39, 41, 43, 45, 47, 49, 51, 53, 55, 57, 59, 60, 62, 64, 66, 68, 70, 72, 74, 76, 78, 80, 82, 84, 86, 88, 90, 92, 94, 96, 98, 100, 102, 104, 106, 108, 110, 112, 114, 116, 118]
>>> import ssd1306
>>> oled=ssd1306.SSD1306_I2C(128,32,i2c,60)
>>> oled.text('ala', 1, 1, 1)
>>> oled.show()
>>>

robert-hh, 2019-12-03:
This many addresses happen often if the pull-up resistors are missing. The Pyboard has built-in pull-up resistors.

bgolab, 2019-12-03:
Yes, I was suspecting the pull ups problem at the beginning.

I use chinese clone of the pyboard. Need to check it.
The PCA10028 clone I use (actually Waveshare BLE400) has no pull-up resistors on the board. 
It seems (looking at PCB) that the OLED (the I2C slave) has such resistors.
I also tested FRAM module with pull ups. And saw the same results.

I use these I2C modules for my regular I2C development where I implemented scanner. And all worked fine.

Anyway - the I2C communication is reliable here. I was misled by the scan.

glennrub, 2019-12-03:
@bgolab, i did a test run on a micro:bit which i expected to work. I'm able to reproduce some issue. I would not be surprised if they correlate.
```
MicroPython v1.11-611-g7f24c2977 on 2019-12-03; micro:bit with NRF51822
Type "help()" for more information.
>>> from machine import Pin, I2C
>>> i = I2C(0, scl=Pin(0), sda=Pin(30))
>>> i.scan()
[14, 29, 90]
>>> i.scan()
[14, 24, 29, 111]
>>> i.scan()
[14, 29]
```
Some quick tests shows that the nrfx driver reports false results (success where it should not). I'll do some more investigation of the nrfx twi driver to see if it leads somewhere.

bgolab, 2019-12-04:
Some maybe useful info:
- my nrf51 board reported addresses repeatable  - always the same set
- I am not 100% but I believe I got similar results on ESP32 (I have no logs)
- pyboard sometimes reported two addresses 

So it may be indication of timing problem  - referring to your results where you have random sets of data