micropython/micropython #5376: Empty op name printed when missing from mp_binary_op_method_name

Jongy, 2019-12-04:
With `MICROPY_PY_ALL_SPECIAL_METHODS =0`,

```
>>> 1 + 1
2
>>> 1 + 1 / 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for  : 'int', 'int'
>>> 
```

You can see the print is a bit broken and it makes you question whether something deeper has gone wrong. I think it would be better to print *something*, like "unknown" or "unknown op".

For example, `[0 ... MP_BINARY_OP_NUM_RUNTIME-1] = MP_QSTR_unknown,`  as the first line of `mp_binary_op_method_name` does the trick. It basically adds just this single "unknown" qstr.

Jongy, 2019-12-04:
As per https://gcc.gnu.org/onlinedocs/gcc/Designated-Inits.html, the range initializer trick is a GNU extension so it won't be available on msvc, I guess.

dpgeorge, 2019-12-04:
> You can see the print is a bit broken and it makes you question whether something deeper has gone wrong.

Yeah, this behaviour has been around since the beginning.  It is a little unsettling seeing the broken error message.

> the range initializer trick is a GNU extension so it won't be available on msvc, I guess.

Yes.  You can see `py/vmentrytable.h` uses this trick but it's not usable in strict c99 mode.

Also, using this scheme might break some code because then a binop/unop lookup on a user class would lookup `unknown` for operations that don't have a corresponding dunder name defined...