micropython/micropython #5378: [ESP32] Partition.find(label='x') not supported.

nevercast, 2019-12-04:
When trying to find a partition by label only, on ESP32, the returned list is empty. 

```
>>> Partition.find(label='ident')
[]
>>> Partition.find(type=0x5A)
[<Partition type=90, subtype=0, address=36864, size=24576, label=ident, encrypted=0>]
```

If its not too much effort, it would be nice to just give the label name and find the partition.

jimmo, 2019-12-04:
`Partition.find` is used (and works) in `esp32/modules/flashbdev.py`:

```python
bdev = Partition.find(Partition.TYPE_DATA, label='vfs')
```

I think you need to specify the partition type?

nevercast, 2019-12-04:
For refernce, here is @dpgeorge original comment on the functionality:
https://github.com/micropython/micropython/pull/4910#issuecomment-522203967

> To construct a Partition you now either pass in a label, or a constant, like so:

Passing just the label also does not work, 
```
>>> Partition('ident')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```

@jimmo 
Yes, passing both parts does work, but it seems that label should just work by itself also, and it does not. 

```
>>> Partition.find(type=0x5A, label='ident')
[<Partition type=90, subtype=0, address=36864, size=24576, label=ident, encrypted=0>]
```

nevercast, 2019-12-04:
Seems this is probably an issue with Espressif, since MicroPython just hands off the search to ESP IDF https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/storage/spi_flash.html#_CPPv418esp_partition_find20esp_partition_type_t23esp_partition_subtype_tPKc

dpgeorge, 2019-12-04:
As mentioned in https://github.com/micropython/micropython/pull/4910#issuecomment-522203967 there are 2 ways to construct a partition object:
1. via constructor, then it accepts an integer constant or string identifier, eg `Partition('fatfs')`
1. via `Partition.find()` for more advanced use and full control over the search

Doing `Partition('ident')` should just work if you have "ident" defined in partitions.csv.  Eg `Partition('vfs')` works on the main firmware.

nevercast, 2019-12-04:
Here is my partition table:
```
ident,    0x5A, 0x00,    0x9000,  0x6000
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x180000,
```

Note that `Partition('factory')` works, as well as phy_init, but `Partition('ident')` does not. 
```
>>> Partition('factory')
<Partition type=0, subtype=0, address=65536, size=1572864, label=factory, encrypted=0>
>>> Partition('phy_init')
<Partition type=1, subtype=1, address=61440, size=4096, label=phy_init, encrypted=0>
>>> Partition('ident')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```

But, ident is found by type, as above, 

```
>>> Partition.find(type=0x5A)
[<Partition type=90, subtype=0, address=36864, size=24576, label=ident, encrypted=0>]
```

nevercast, 2019-12-04:
The IDF docs does have this one line:
`type: Partition type, one of esp_partition_type_t values`

For which, `esp_partition_type_t` values are:
https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/storage/spi_flash.html#_CPPv420esp_partition_type_t

> ESP_PARTITION_TYPE_APP = 0x00
> Application partition type.
> 
> ESP_PARTITION_TYPE_DATA = 0x01
> Data partition type.

Perhaps custom partition types are not supported when searching by label.

nevercast, 2019-12-04:
Further documentation on custom partitions: https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/partition-tables.html#type-field

The boot loader ignores any type other than 0x00 and 0x01. While I was aware of this and found this satisfactory. I didn't expect other features of the partition API to ignore it. 

Either way, I no longer think this is a MicroPython bug and I'm happy to close this. Perhaps the documentation could do with a bump. Maybe I should poke Espressif on this.

dpgeorge, 2019-12-04:
> Perhaps custom partition types are not supported when searching by label.

Right, that's the problem.  The IDF doesn't (AFAIK) provide a way to search all partition types for a given identifier.  You must know the type to search for it.  The MicroPython bindings search both TYPE_APP and TYPE_DATA (in that order) when given a string, and fails if the string is not found under one of those types.

I don't think it's sensible for MicroPython to search all types (eg a for loop over 0x00-0xff).

I guess the point is you can define the same label more than once for different types.

nevercast, 2019-12-04:
That sounds sensible. Thanks!

nevercast, 2020-06-15:
**Conclusion**: You cannot search only by a label if the partition type is not APP or DATA, this is an ESP-IDF limitation. If you're searching for a partition by name and it is not either APP or DATA, then you must also explictly specify the type, eg:
```Partition.find(type=0x5A, label='ident')
```

**wont-fix**

nevercast, 2020-06-15:
**Conclusion**: You cannot search only by a label if the partition type is not APP or DATA, this is an ESP-IDF limitation. If you're searching for a partition by name and it is not either APP or DATA, then you must also explictly specify the type, eg:
```
Partition.find(type=0x5A, label='ident')
```

**wont-fix**