micropython/micropython #5382: (In)Equality operator is constrained to returning bool

nickovs, 2019-12-04:
In CPython the equality `==` and inequality `!=` operators can return any sort of object. This is used in systems such as [`numpy`](https://numpy.org/) to allow element-wise comparison of arrays. In the current Micropython [implementation](https://github.com/micropython/micropython/blob/7aeafe2ae9f16af74d22604b3090641a664b2da2/py/runtime.c#L324) the (in)equality operators are constrained to return `bool` values, which prevents this sort of functionality from being implemented. Furthermore the current implementation [short-cuts](https://github.com/micropython/micropython/blob/7aeafe2ae9f16af74d22604b3090641a664b2da2/py/obj.c#L162) comparison of an object with itself, which will also prevent consistent implementation of classes that return non-bool types from equality comparison even if any type of result is allowed.

These behaviours are both inconsistent with CPython and also inconsistent with the implementation of the magnitude comparison operators (`>`, `>=`, `<` and `<=`), which can return any type and have no short-cut.

It would be helpful to be able to implement equality comparison that is compatible with CPython.

dpgeorge, 2019-12-05:
Yes, I noticed this inconsistency recently as well, in the context of non-reflexive classes.  For example:
```python
class A:
    def __eq__(self, other):
        return False
a = A()
print(a == a)
```
This returns `False` in CPython but `True` in uPy.

I would be good to fix this properly.  It requires a bit of an overhaul of the way (in)equality is computed.

nickovs, 2019-12-05:
It would be interesting to try taking the pointer equality shortcut out (or replace it with something that only checks when they are really short ints) and see if it makes a meaning performance difference.

nickovs, 2019-12-09:
Related both to the inconsistency that you spotted and the one I mentioned, I notice that the `__ne__` special method can only be called manually (using `getattr()` or similar) and it's more or less impossible to trigger the `MP_BINARY_OP_NOT_EQUAL` op code in a native class's `binary_op` function. The test cases in `tests/basics/special_methods.py` conspicuously fail to test this special method.

It seems like there are only a couple of places that need to be reworked to fix this. We can make the existing `mp_obj_equal()` function be a wrapper of `mp_obj_is_true()` around a call to a new function that handles both equality and inequality cases, supporting various short-cuts before trying the binary operator and falling back to pointer equality as a last resort, and then just replace the code in `mp_binary_op()` that handles (in)equality with calls to this new function. I will have a go at implementing this so that we can see what the size and performance impacts are.

dpgeorge, 2020-01-30:
Fixed by #5593, commits c3450effd4c3a402eeccf44a84a05ef4b36d69a0 through c96a2f636b48b065e8404af6d67fbae5986fd34a

closed 2020-01-30 by dpgeorge