micropython/micropython #5384: Two argument form of iter() is not implemented.

hardkrash, 2019-12-05:
The builtin `iter(callable, sentinel)` from pep [234](https://www.python.org/dev/peps/pep-0234/#python-api-specification) is not implemented.

The use case is something like this.

```python
with open('file.bin', 'rb') as f:
    for chunk in iter(lambda: f.read(128), b''):
        process_chunk(chunk)
```
The two parameter form allows an easy way to make a callable into an iterable by using a sentinel.

It allows for replacing the do while workarounds with a simple iterator.

```python
with open('file.bin', 'rb') as f:
    chunk = f.read(128)
    while chunk != b'':
        process_chunk(chunk)
        chunk = f.read(128)
```

Credit to this stack overflow answer from [jfs](https://stackoverflow.com/users/4279/jfs)
https://stackoverflow.com/a/20014805

mikewadsten, 2019-12-05:
Sort of related to #4400.

dpgeorge, 2019-12-06:
Yes, this is simply not implemented, mainly because the same thing can be achieved in pure Python.  As a workaround the following can be used:
```python
def iter2(c, s):
    while True:
        v = c()
        if v == s:
            raise StopIteration
        yield v
```
(This would be simpler with the walrus operator!)

hardkrash, 2019-12-17:
While I agree on the fact that this can be worked around in many ways.  it is unfortunate that this is not explicitly called out in the documentation as a difference from python's core syntax.

This would be a compatibility wrapper needed to make the two argument form of iter work without changing source code.

`from compat import iter`

Where compat.py has something like this.

```python
S = some_singleton
_orig_iter = iter
def _compat_iter(c, s=S):
    if s is S:
        for x in _orig_iter(c):
            yield x
    else:
        while True:
            v = c()
            if v == s:
                raise StopIteration
            yield v

iter = _compat_iter
```

Would adding 2 argument form to the builtin iter function be an acceptable patch?

dpgeorge, 2019-12-18:
>  it is unfortunate that this is not explicitly called out in the documentation as a difference from python's core syntax.

It's not really a syntax difference, rather the fact that MicroPython implements a subset of the Python builtins.  It could be added to the docs via `tests/cpydiff` (which automatically generates differences for the docs).

> Would adding 2 argument form to the builtin iter function be an acceptable patch?

Implemented in C I think it'd be a large patch and therefore unlikely to be made part of default MicroPython builds.  It could be made optional like the 2-arg form of `next()`, but I'm just not sure how generally useful it would be (it hasn't been requested so far).

dpgeorge, 2021-10-20:
This would be a good candidate for #5025.

projectgus, 2024-08-28:
Labelling as `docs`, as I think the most sensible to do in the short term is to add this to the documented list of CPython differences.

symbioquine, 2025-03-06:
I was also surprised to discover `next` not taking a second default/sentinel parameter - and not finding any mention of this in https://docs.micropython.org/en/latest/genrst/index.html#micropython-differences-from-cpython

Naively, I would have expected to see a section in the docs on differences in "builtin methods" above/below the section on differences in "builtin types".

[image]

Again naively, this breaks my assumptions about MicroPython a bit because it seems surprising that the logic to support the second parameter would increase the size noticeably, nor present any performance cost. Yes, it's easy to work-around, but I wish I didn't need to consider this among the things constraining code's "portability to MicroPython".

stinos, 2025-03-07:
Note that the 2-argument `next` is implemented, guarded by `MICROPY_PY_BUILTINS_NEXT2`, see https://github.com/micropython/micropython/commit/42863830be2d19c7dcdf7ccf1fa66168b1bdc13a

symbioquine, 2025-03-07:
Okay, looking a bit harder, I actually found the relevant documentation: https://docs.micropython.org/en/latest/genrst/modules.html#second-argument-to-next-is-not-implemented

So I guess that documentation probably needs updating to clarify that it's now not available on *some* boards in the default builds and whether it is available is controlled by that compile time flag?

stinos, 2025-03-10:
Yeah I don't think there's a real policy now for 'conditionally different from CPython'. There are multiple other occurrences in these docs like that.

dpgeorge, 2025-03-10:
Looking at this again after many years, I see that the difference in implementation between 1-arg `next` and 2-arg `next` is very small.

I made a comment long ago:

> Implemented in C I think it'd be a large patch and therefore unlikely to be made part of default MicroPython builds.

I think my reasoning back then was because I was thinking that the implementation would need a NLR block to catch the stop iteration.  But it doesn't anymore because we have `mp_iternext_allow_raise` and `mp_iternext`.

In my opinion the 2-argument form of `next()` should be enabled at a much earlier feature level, like `MICROPY_CONFIG_ROM_LEVEL_BASIC_FEATURES`.  Or even just unconditionally replace the 1-arg form.

dpgeorge, 2025-03-11:
See #16902.

dpgeorge, 2025-03-27:
Now implemented.

closed 2025-03-27 by dpgeorge