micropython/micropython #5385: pyboard v1.1 not recognozed by Windows 10

bgolab, 2019-12-05:
Hi,
Windows 10 64bit. 

The story:
1. The pyboard v1.1 worked fine with the 1.9.3 build (the image for the v1.1 board).

2. Accidentally I flashed the board with WRONG 1.11 version targeted for pybv1.0 (pybv10-20190529-v1.11.dfu)  - and the board WAS RECOGNIZED (both as a disk & com)

3. Once I discovered the mistake I flashed the board with right sw (pybv11-20190529-v1.11.dfu) - the board has NOT been recognized since then (both as a disk & com are NOT recognized). Tested 1.9.3, 1.10, 1.11

4.  Connected the board to Windows 7 64 bits - the board is still NOT recognized both as a disk & com

5. Just for a test I flashed the board with the WRONG image again (pybv10-20190529-v1.11.dfu) and the board is recognized correctly as a disk & com.

----
It looks like the hw is OK.
It looks like the STM32 firmware in the v1.1 version is somewhat broken.
The mystery is why this worked correctly earlier (before wrongly switching to v1.0 firmware).

Any idea what is going on?

bgolab, 2019-12-05:
I would like to add that the board is fully accessible not only through the DFU but also using SWD programmer (st-link, segger). So I do not suspect any hardware problem.

bgolab, 2019-12-05:
After dwelling on scheme differences between v1.0 and v1.1 I figured out that the main difference is the STM32 MCU and xtal speed (8 vs 12MHz).
So decide to try the PyB v3 as it uses the STM32F405 (the same as v1.1) and *MHz clock (the same as v1.0).

And it works - at least I am able to to turn on the LED.
 This is surprising - until today I have no idea about PyB v3.

dpgeorge, 2019-12-06:
This is strange indeed.  If USB is working then it must be using the correct xtal MHz setting.

Do you have an official PYBv1.1?

> I would like to add that the board is fully accessible not only through the DFU but also using SWD programmer (st-link, segger).

The DFU mode does not depend on the xtal frequency, it should always work if the hardware is not faulty.

bgolab, 2019-12-06:
It's a pyb clone. The only clone I ever had. 
I went through the file dates. I started using this board in Apr, 2018 (uP 1.9.3). An have several scripts with this date. And the schematic file for the v1.1. None of these scripts works now (pin naming is different),

I do not think we can do anything about this now because it is so weird and I cannot explain it. 
From the board perspective it looks like the wrong xtal is used (image does not match the hw).

I am involved in STM32 C development for 10 years, and I am rather strict if it comes to change management, hw documentation, etc - but this time it looks like I did messed something.

dpgeorge, 2019-12-06:
> It's a pyb clone. The only clone I ever had.

In that case it's hard to tell where the issue lies.  In particular it could be that this board shipped with firmware for a v1.0 board, or modified v1.1 firmware to use a 8MHz xtal.

Otherwise it's not really a bug in software, and not official hardware, so I'll close this issue.

closed 2019-12-06 by dpgeorge