micropython/micropython #5387: ESP8266 has no littlefs support?

jedie, 2019-12-06:
I used the code snippted from: http://docs.micropython.org/en/latest/reference/filesystem.html#littlefs to convert the filesystem.

But get these errors:
```
os.VfsLfs.mkfs(flashbdev.bdev) # AttributeError: 'module' object has no attribute 'VfsLfs'
os.VfsLfs2.mkfs(flashbdev.bdev) # AttributeError: 'module' object has no attribute 'VfsLfs2'
```

Tested on/with:
```
ESP module with ESP8266 2.2.0-dev(9422289)
MicroPython v1.11-8-g48dcbbe60 on 2019-05-29
```

mzdaniel, 2019-12-06:
Although work in LittleFS spans for some time, general availability on Micropython ports is pretty recent. I think the docs refer to the new to be released Micropython  v1.12.

mzdaniel, 2019-12-06:
This [table](http://docs.micropython.org/en/latest/reference/filesystem.html#filesystems) explains your findings too.

jedie, 2019-12-06:
I tried latest initial OTA **esp8266-ota-20191206-v1.11-624-g210d05328.bin** and get the same: _AttributeError: 'module' object has no attribute 'VfsLfs2'_

peterhinch, 2019-12-07:
Why is it not enabled by default on ESP8266?

jimmo, 2019-12-07:
> I tried latest initial OTA **esp8266-ota-20191206-v1.11-624-g210d05328.bin** and get the same: _AttributeError: 'module' object has no attribute 'VfsLfs2'_
> 
> EDIT: Same with non OTA image: **esp8266-20191206-v1.11-624-g210d05328.bin**

@jedie please see the table linked by @mzdaniel above. The ESP8266 does not include LFS support by default, but support can be optionally enabled in a custom firmware build. I think you just need to add `MICROPY_VFS_LFS2=1` to your `make` command (or add it to your `mpconfigboard.mk` if you have a custom board definition).

closed 2019-12-07 by jimmo

peterhinch, 2019-12-08:
True. But you should also adapt inisetup.py so that, on first boot after erasing flash, the filesystem is created:
```python
    uos.VfsLfs2.mkfs(bdev)
    vfs = uos.VfsLfs2(bdev)
#    uos.VfsFat.mkfs(bdev)
#    vfs = uos.VfsFat(bdev)
```

dpgeorge, 2019-12-09:
> Why is it not enabled by default on ESP8266?

Because it doesn't really have enough code space (it does just have enough but then there's no room left for user frozen code).

Ideally in the next release we can switch to littlefs by default, and also increase available code space to the maximum.

andrewleech, 2020-01-22:
My unofficial micropython CI now has LFS builds for Esp8266 (micropython master builds only at the time of writing) for anyone who isn't set up to compile their own.

https://alelec.gitlab.io/micropython_ci/