micropython/micropython #5389: docs contains information that are not exist in official builds

jedie, 2019-12-06:
As https://github.com/micropython/micropython/issues/5387 shows, there are contents in the documentation that are not yet available in the official downloads.

These should be marked in the docs.

cwalther, 2019-12-06:
Have you tried switching from “latest” to “v1.11” at the bottom of the sidebar? I assume “latest” refers to the current development version, not to the last stable release.
Maybe http://docs.micropython.org/ shouldn’t redirect to “latest” but to the stable release? I have no idea if Read the Docs allows that though.

jedie, 2019-12-07:
Ah! You're right. e.g.:

"latest" has Filesystem/Littlefs: http://docs.micropython.org/en/latest/reference/filesystem.html#littlefs (See also: https://github.com/micropython/micropython/issues/5387 )
and "v1.11" doesn't contain the new "Filesystem" section: http://docs.micropython.org/en/v1.11/reference/index.html

But yet: I Think http://docs.micropython.org/ should redirect to stable release version.

peterhinch, 2019-12-08:
So long as those of us who like to live at the bleeding edge can still access "latest" :)

dpgeorge, 2019-12-09:
I agree it'd make sense to make clearer the distinction between "latest" and "most recent stable", and make it easier to switch between the two.  At the moment the front page has the same intro for both of these, namely "Welcome! This is the documentation for MicroPython v1.11".

I like how Django does it: defaults to latest stable release https://docs.djangoproject.com/en/3.0/, has a warning at the top for very old releases https://docs.djangoproject.com/en/1.7/ and a warning for bleeding edge development https://docs.djangoproject.com/en/dev/