micropython/micropython #5390: 'uasyncio' missed in ESP8266 daily build v1.11-624

jedie, 2019-12-06:
I switched from stable ESP8266 firmware **esp8266-20190529-v1.11.bin** to the last daily build **esp8266-20191206-v1.11-624-g210d05328.bin** and 'uasyncio' missed.

jimmo, 2019-12-07:
I don't know the historical reason, but the stable builds use the release manifest (`ports/esp8266/manifest_release.py` which adds a bunch of extra things including umqtt, urequests, uasyncio), whereas the nightly builds use the default manifest (`ports/esp8266/manifest.py`).

jedie, 2019-12-07:
OK, that's the reason. But it doesn't make any sense and should be changed, right?

dpgeorge, 2019-12-09:
> I don't know the historical reason

I guess it traces back to the MicroPython esp8266 Kickstarter, where early release builds were issued before the source code was fully available, and also the idea was to pack in some examples scripts that could be used out-of-the-box to show off its capabilities.

But it might be a good idea to change this behaviour moving forward, to simplify things for users and for the release process.  Also needs to be changed then for esp32, which has a separate `manifest_release.py`

jedie, 2019-12-20:
Current, official release **esp8266-20191220-v1.12.bin** from http://micropython.org/download#esp8266 has still no littlefs2 support:
```
MicroPython v1.12 on 2019-12-20; ESP module with ESP8266
Type "help()" for more information.
>>> import os
>>> dir(os)
['__class__', '__name__', 'remove', 'VfsFat', 'chdir', 'dupterm', 'dupterm_notify', 'getcwd', 'ilistdir', 'listdir', 'mkdir', 'mount', 'rename', 'rmdir', 'stat', 'statvfs', 'umount', 'uname', 'urandom']
```

Josverl, 2020-01-01:
thansk for the info . I landed here on a quest to understand which scripts are frozen into which ports/boards and could not find any info regarding `manifest_release.py`. 

what is the plan moving forward ?

dpgeorge, 2020-09-17:
See #6434 which removes the release-specific manifest on the esp8266.

dpgeorge, 2020-09-18:
Fixed by b28758054b586fe41f56e462856d89f28c5e3f6a

closed 2020-09-18 by dpgeorge