micropython/micropython #5392: mpy-cross bytecode optimizations level ?!?

jedie, 2019-12-07:
The mpy-cross help or https://github.com/micropython/micropython/blob/master/mpy-cross/README.md does not describe which optimizations level exist. And what the levels do.

I notice that the optimizations level, set with `-O`, changes the information in tracebacks. At higher levels, for example, the code line number is always 0.

Is there more detailed information about this somewhere?

I also ask here:

- https://gitlab.com/alelec/mpy_cross/issues/5
- https://forum.micropython.org/viewtopic.php?f=2&t=7350

jimmo, 2019-12-07:
See #5393

The optimisation level is the same as setting `micropython.opt_level` which is documented at http://docs.micropython.org/en/latest/library/micropython.html#micropython.opt_level

jedie, 2019-12-07:
Thanks! I will close this fpr #5393

jedie, 2019-12-07:
I looked at: http://docs.micropython.org/en/latest/library/micropython.html#micropython.opt_level

reopen this issues, because I miss the highest value. I see only: 

> levels 3 and higher

So there must exists a level 4 ?

And another question: What's about DocStrings and __doc__: Will they be always striped?

jimmo, 2019-12-07:
> So there must exists a level 4 ?

Any optimisation behaviour is enabled at the given level, and any higher levels. But in reality it's just a number... so I guess a level 4 could be added in the future (which included all O1, O2, O3 optimisations) but currently the highest level used is `3`. If you grep the codebase for `mp_optimise_value` you can see them.

> And another question: What's about DocStrings and **doc**: Will they be always striped?

The behaviour of docstrings (and the `__doc__` attribute) is controlled by the `MICROPY_ENABLE_DOC_STRING` configuration option (and it's disabled by default).

So yes, docstrings are always stripped regardless of the optimisation flag (this applies to both mpy-cross and the on-board compiler).

closed 2019-12-07 by jimmo