micropython/micropython #5394: How to access modules files?

RJSDevel, 2019-12-07:
I want to flash modules with html files together, but I can't read them.
e.g list 
_boot.py
public/index.html

I modified makemanifest.py for pass other files types as path, I cheked existing it in the firmware.elf, and it exist, but I cant read it for send by http

May be exist way for read those files or what sorces can I modify for get this possibility

jimmo, 2019-12-07:
This is possibly a longer-term feature for the manifest system, but not currently implemented. At the moment it only generates frozen modules (with optional compiling to bytecode first).

The code that loads modules at runtime (i.e. the import statement) then preferentially finds these frozen modules before looking on the filesystem. However the filesystem itself is totally separate and not part of the firmware image.

It is possible however to construct a filesystem and append it to the firmware image. See for example https://forum.micropython.org/viewtopic.php?f=16&t=6677 for some general ideas but also at the end of the thread there's an ESP8266-specific example of doing this all automatically.

peterhinch, 2019-12-08:
[This utility](https://github.com/peterhinch/micropython_data_to_py) enables an arbitrary file to be converted to Python source which can be frozen.

RJSDevel, 2019-12-09:
> This is possibly a longer-term feature for the manifest system, but not currently implemented. At the moment it only generates frozen modules (with optional compiling to bytecode first).
>
> The code [...]

I think this is right way. Thank you!

TravisJoe, 2019-12-19:
I have done something based on Peter's solution which has worked quite well for me. Plus easily do dynamically build and move into my frozen file directory when building the firmware.

jedie, 2020-01-08:
> It is possible however to construct a filesystem and append it to the firmware image. See for example https://forum.micropython.org/viewtopic.php?f=16&t=6677 for some general ideas but also at the end of the thread there's an ESP8266-specific example of doing this all automatically.

That sounds as the best way. Would be great if this is doable in the normal micropython build pipeline.

It would also be good not only not to transfer non-py files on the device, but also to avoid having to create the filesystem on the first startup e.g. with **inisetup.py**.

I see the advantages over Peter's solution in the fact that you can open the files normally. So it is also possible to change the files, like all normal files.

peterhinch, 2020-01-08:
> I see the advantages over Peter's solution in the fact that you can open the files normally. So it is also possible to change the files, like all normal files.

That is true, and there are applications for a filesystem populated at build time. The approaches have different benefits and drawbacks.

My approach has the following benefits, especially if you need random access e.g. for fonts:
 1. If the frozen Python code supplies a memoryview into a bytes object, RAM usage can be almost zero (mutable objects must be located in RAM).
 2. Accessing a memoryview slice into a bytes object in flash is much faster than a seek into a random access file.

The immutability of frozen bytecode is therefore both a benefit and a limitation.