micropython/micropython #5398: open() implementation differs from CPython and is creating a file with different name than specified

panekk, 2019-12-08:
During making some experiments, I noticed that this code surprisingly works:

> f = open('test abc\ndef.txt', 'w')
> f.write('some data')
> f.close()

and a `test abc` file is created.

In CPython (3.7) running the first line leads to `OSError: [Errno 22] Invalid argument` error.

I tested it on ESP8266 with MicroPython v1.11-8-g48dcbbe60.

mzdaniel, 2019-12-08:
It is important to realize micropython cannot possible replicate cpython behavior one by one, especially on edge cases like this where there is an intentional invalid argument. On a desktop machine, python3.8 takes at least of 10 megabytes of ram just to launch (and this does not include the OS). ESP8266 has 64 kilobytes of total ram for doing everything (quite a limited space for today's standards), almost 3 order of magnitude smaller.  Don't you rather use the resources available to do something more productive than printing a pretty error message?

panekk, 2019-12-08:
It's not about printing a pretty message. This behavior is clearly invalid, the function doesn't do what it's supposed to do, doesn't report ANY error and user doesn't know about it, until checking what files are in fact in the directory.

Small amount of memory isn't a excuse for such behavior. Also, there's error when I'm trying to create `\ntest.txt`, so probably there's some validation, but it isn't good enough. I'm going to investigate that when I finally compile the project.

jimmo, 2019-12-08:
Note that this may be host-OS (and most likely host-filesystem) dependent.

For example, on the Unix port, the above snippet gives the same behavior between MicroPython and CPython. (Both running on Linux with ext4).

I agree though that if on ESP8266 (using FatFS), that if `open('something', 'w')` succeeds but creates a file name different to `'something'` (which sounds like what you're describing), then that's a bug. However, the bug here might be in FatFS, not in MicroPython?

panekk, 2019-12-09:
Looks like MicroPython's code doesn't validate filename before passing it to FatFS, so you're probably right!

I'll investigate it further tomorrow. MicroPython uses version R0.09b of FatFS and the newest one is R0.14, so it may be fixed already.

jimmo, 2019-12-09:
MicroPython uses FatFS 0.13c -- via https://github.com/micropython/oofatfs