micropython/micropython #5400: is that a bug in sdcard auto mount during boot

metadata1984, 2019-12-09:
there is statment in ports/stm32/main.c

if (!mounted_flash || mp_vfs_import_stat("SKIPSD") == MP_IMPORT_STAT_FILE) {

and in my custom board . it can not auto mount sd card during boot

than i change to 

if (!mounted_flash || mp_vfs_import_stat("SKIPSD") == MP_IMPORT_STAT_NO_EXIST) {

it is ok.

jimmo, 2019-12-09:
Thanks! Looks like in the update a few days ago to change from using fatfs directly to using the vfs the condition was inverted.

before:
```
if (!mounted_flash || f_stat(&fs_user_mount_flash.fatfs, "/SKIPSD", NULL) != FR_OK) {
```

after:
```
if (!mounted_flash || mp_vfs_import_stat("SKIPSD") == MP_IMPORT_STAT_FILE) {
```

Sent PR #5401 to fix.

dpgeorge, 2019-12-09:
Thanks for the report!  Fixed by b76f0a73bc31d1c8871ff00d2f93b5805b3b6315

closed 2019-12-09 by dpgeorge