micropython/micropython #5402: [NUCLEO_F767ZI] Internal Flash erased after every reset

rolandvs, 2019-12-09:
When initialising the Flash drive: hold user button, press reset, wait for two LEDs on, release user button. It will generate a file system on a volume with name "PYBFLASH" and the set of default files. Then after hitting the reset button and the drive re-appears the disk has the name "NO NAME" and is empty.

dpgeorge, 2019-12-10:
I cannot reproduce this problem.  Using Linux, following the above process, the USB MSC drive is mountable after each reset and contains the factory files.

rolandvs, 2019-12-11:
@dpgeorge I have bought another F767 board to see if the board in question is to be replaced and buried :-)

dpgeorge, 2019-12-11:
It may be that the flash is worn.  You could try a mass erase.

rolandvs, 2019-12-11:
Solved: the option bits of the F767 have a flag called nDBANK that needs to be set %^&*. The 2MByte Flash was divided in two 1MByte parts.

closed 2019-12-16 by dpgeorge