micropython/micropython #5405: Move esp32 REPL to a different UART

lerouxb, 2019-12-09:
Hi.

I'm trying to change the ESP32 REPL to use UART2 rather than UART0. I'm using an Adafruit Feather Huzzah32 and I want to be able to use the REPL over the feather's RX/16 and TX/17 pins rather than the USB port. I believe this is UART2.

I've tried the following:

```
diff --git a/ports/esp32/uart.c b/ports/esp32/uart.c
index 10a4ba462..1de31dc0c 100644
--- a/ports/esp32/uart.c
+++ b/ports/esp32/uart.c
@@ -37,13 +37,13 @@ STATIC void uart_irq_handler(void *arg);

 void uart_init(void) {
     uart_isr_handle_t handle;
-    uart_isr_register(UART_NUM_0, uart_irq_handler, NULL, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM, &handle);
-    uart_enable_rx_intr(UART_NUM_0);
+    uart_isr_register(UART_NUM_2, uart_irq_handler, NULL, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM, &handle);
+    uart_enable_rx_intr(UART_NUM_2);
 }

 // all code executed in ISR must be in IRAM, and any const data must be in DRAM
 STATIC void IRAM_ATTR uart_irq_handler(void *arg) {
-    volatile uart_dev_t *uart = &UART0;
+    volatile uart_dev_t *uart = &UART2;
     uart->int_clr.rxfifo_full = 1;
     uart->int_clr.frm_err = 1;
     uart->int_clr.rxfifo_tout = 1;
```

But this doesn't seem to be enough. I've searched all over this codebase and cannot figure out how this fits together. Is there something obvious I'm missing?

MrSurly, 2019-12-09:
Just a guess here, but I suspect you need to enable those pins to be UART pins.

Per [this documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/uart.html#setting-communication-pins).

I'm nearly certain that UART0 is fully setup on boot, but I don't think the others are.

lerouxb, 2019-12-10:
Hmm I've also found this while searching through esp-idf:

```
components/esp32/Kconfig
559:            Due of a ROM bug, UART2 is not supported for console output
```

Wondering how far that goes, because looking through here, a lot of things only mention UART0 and UART1, but not UART2: https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp32s2beta/rom/uart.h

Maybe I should change the repl code to use other routines? I see this code uses uart_tx_one_char() which doesn't specify which uart you mean which is what I guess uart_tx_switch() is far and it probably defaults to 0. But uart_tx_switch only supports 0 or 1. Whereas the code in machine_uart.c uses things like uart_write_bytes which does seem to support 0, 1 or 2.

Perhaps I should change the code to use that?

lerouxb, 2019-12-10:
Hmm and if the rom code did support uart2, then it looks like this is the kind of code that would be necessary to configure those pins: https://github.com/espressif/esp-idf/blob/a45e9985344575a80acb1dc9c12e7bec4d8af401/components/bootloader_support/src/bootloader_init.c#L389-L419

smurfix, 2021-06-20:
Hmm. Looks like the UART should use a separate OS thread to transfer the data from the UART to the console buffer. The current solution mangles their interrupt routine, which is a bit of a hack.