micropython/micropython #5406: Any plan to support classic bluetooth in ESP32?

jxltom, 2019-12-10:
It looksl like bluetooth classic is supported in ESP32 by espressif in https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/classic_bt.html#. Any plan to support it in micropython?

jimmo, 2019-12-10:
It would be good to support it (and on STM32 too), but I don't think it's on the roadmap. If people are keen to work on it then it would be good to discuss a way forward.

Some notes/considerations about BT classic:

 - On ESP32 BLE we use the NimBLE stack (to re-use the work done for STM32). BT classic requires Bluedroid, and the same firmware build can't have both enabled. So some thought would need to go into avoiding having to have even more build configurations. Bluedroid supports LE too though.

 - The API for BT classic is likely going to be a lot more complicated. I imagine it might be easier to focus on some specific use cases (e.g. BT profiles) and build APIs for them instead, for example A2DP audio source/sink is probably the top of my list. What use cases did you have in mind?

jxltom, 2019-12-10:
Thanks for updating the status and the delightful suggestions. Sounds like there is still a long way to go, but look forward to it.

HFP source/client, especially client, might also be the use case.

montyx99, 2020-05-09:
I'd like to start with a simple UART to Bluetooth connection. The PCs can attach only classic Bluetooth connection as a raw serial port. Can you give me any advice to start to implement this?

zeilenschubser, 2020-05-09:
@montyx99 https://forum.micropython.org/viewtopic.php?t=8309

emard, 2021-02-20:
I would like to connect ESP32 with bluetooth GPS,
which provides bluetooth classic serial port as slave device.
(esp32 initiates connection)

I have tested it with arduino esp32 bluetoothserial
and it works perfectly, more reliable than connecting
from linux.

so I'd expect micropython could connect also.

jimmo, 2021-02-22:
> so I'd expect micropython could connect also.

@emard Yes the ESP32 hardware supports Bluetooth Classic, however MicroPython does not expose bindings for it. At this stage we've focused on just exposing the bindings for the Bluetooth Low Energy stack.

avsaase, 2021-10-16:
I want to express my support for this feature request. There exist some very cheap BT modules like the HC-05 and HC-06 that are perfect for simple projects but don't support BLE. It would be great if micropython supported Classic BT so I don't have to use C++.