micropython/micropython #5413: Does it support smartlink and ota for the esp32?

saviourxx, 2019-12-12:
Does it support smartlink and ota for the esp32? 
I use micropython on esp32 for smartlock，and ready to trial production. but those two function is the bit problem for the production.
Thanks a log!

beyonlo, 2019-12-12:
@saviourxx 

The OTA and WiFi MESH are in the next roadmap version of MicroPython https://github.com/micropython/micropython/issues/4821

What is the smartlink feature of ESP32? Please, do you have any url showing the smartlink feature?

saviourxx, 2019-12-13:
Dear @beyonlo:

	Smartlink is also called Smartconfig, https://github.com/lanyiyang/esp8266-smartconfig
	Smartconfig and Airkiss are necessary for any wifi products.
	And when is v2.0 release time?
	Thank you!

jonnor, 2024-09-08:
Smartconfig/smartlink/airkiss is not supported, I believe. Hopefully this can be done as add-on modules.

OTA on ESP32 has several other issues open. Proposing to close this one

closed 2025-03-03 by jonnor