micropython/micropython #5422: machine.rtc feature request: rtc.ticks_ms()

peterhinch, 2019-12-16:
Support `rtc.ticks_ms()`. This would return a value compatible with the port's `utime.ticks_diff()` and `utime.ticks_add()`.

Use case: `uasyncio` uses `utime.ticks_ms()` as its timebase. In a light sleep state this clock stops. However it is possible to replace this clock with one based on the RTC. This enables a class of application which waits, with reduced power consumption, on an event such as a button press or a time period. `uasyncio` continues to run, but spends most of the time in light sleep. I have tested this with custom code on Pyboards. This feature would enable portable code to be written.

FYI:
This works by running a coro which puts the target into light sleep for a period. When it wakes, it allows all pending coros to run once before sleeping again. Coros can alter the period or disable sleeping, enabling dynamic control of power consumption/performance. This works on Pyboards under my fork of `uasyncio`. Porting it to new `uasyncio` is waiting pending a stable version but I can't see any obvious difficulty.

dpgeorge, 2019-12-17:
It might be better to just make `utime.ticks_ms()` work correctly over lightsleep.  On esp32 this already seems to be the case.  On stm32 it could be done either of:
- adjust the tick value upon wake from lightsleep (based on measuring the length of sleep using RTC)
- make `ticks_ms()` use the RTC instead of the CPU systick (RTC is actually more accurate anyway).

peterhinch, 2019-12-17:
That sounds ideal. I do think low power uasyncio support is worthwhile.

> RTC is actually more accurate anyway

On the D-series. The Pyboard 1.x RTC has only 1/256s precision.

dpgeorge, 2019-12-17:
> I do think low power uasyncio support is worthwhile.

In the end it could be that poll/select is the entity that does the low-power sleeping, so it "just works".  Well, still some effort required to see if that would work.

peterhinch, 2019-12-17:
That sounds ideal. Are there circumstances in which an "off" switch might be required for this? Do all platforms support interrupts in lightsleep?