micropython/micropython #5425: Ghost interrupts on GPIO39/Sensor_VN

gmos, 2019-12-16:
I have 4 identical input circuits on GPIO's 39, 34, 35, 27.  All 4 configured to do a call back on rising and falling signal.  Worked flawlesly for months.
Yesterday I have upgraded the SDK and Micropython to the latest available V3 version (fb014155996c7568eb55ab06fdeaf6ff414f0280).
Suddenly I getting regularly spurious interrupts on GPIO 39, perhaps once every second.  The other 3 GPIO's are still ok.
On my scope I see a very short pulses when the interrupts happen. Pin is pulled high with 3k3, and the voltage drops with about 1v a very short while.

Problem reproduces on multiple boards.

After digging into the https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf section 2.1.6 on page 7 the hardware issue is probably explained there.  It has to do with ADC or Hall activity.  They recommend to work around in software.
I'm not using one of the ADC's nor the Hall.

What causes the change in behaviour since the upgrade of Micropython and the SDK?
Is this a bug that will be fixed?

dpgeorge, 2019-12-17:
> What causes the change in behaviour since the upgrade of Micropython and the SDK?

It depends which version you started with, which version was working OK.  But most likely it's due to an update of the IDF, eg due to change in how the WiFi power saving feature works.

> I'm not using one of the ADC's nor the Hall.

SARADC2 is used by the WiFi and also affects GPIO 39 (SENSOR_VN) according to the hardware design guidelines.

> Is this a bug that will be fixed?

The problem sounds like a limitation of the hardware and not something that's easily addressed on the MicroPython side.

gmos, 2019-12-17:
Yes. Also suspected an IDF change.  Well, no interrupts on this input.  The pulse is so short that I still can use it as a level input.
Thanx.

closed 2019-12-17 by gmos