micropython/micropython #5426: ESP32 Makefile error

gmos, 2019-12-16:
ESP32 Makefile errors out with "Makefile:102: *** missing separator.  Stop." if the ESP-IDF hash does not matches the recommended V3 or V4 hash.

IveJ, 2019-12-16:
Hi Issuer,

See problem if any match your err.
https://stackoverflow.com/questions/16931770/makefile4-missing-separator-stop

gmos, 2019-12-16:
Nope it is not a TAB error.  But I have found the problem.  In the makefile $(call to_something) is used to print dome stuff.  But call is meant to call a function that give a (possibly null) result. This result needs to be assigned to something. Which does not happen.

So change:
`$(call print_supported_git_hash)` to 
`dummy = $(call print_supported_git_hash)`  to fix the problem.

It is in twice, around line 70 and 102.

dpgeorge, 2019-12-17:
>  In the Makefile $(call to_something) is used to print some stuff. But call is meant to call a function that give a result, which is now text interpreted as a statement with an error.

This problem has not been seen before.  What version of GNU make do you use?

gmos, 2019-12-17:
GNU Make 4.1
Built for x86_64-pc-linux-gnu

jimmo, 2019-12-17:
Interesting... this is indeed a make 4.2 "fix". See http://savannah.gnu.org/bugs/index.php?46400 and http://savannah.gnu.org/bugs/index.php?46995

Make 4.2 was released 3.5 years ago... any chance you can upgrade? Otherwise I guess there's no harm in adding your workaround.

gmos, 2019-12-17:
Better check my workaround then in de newest version.

Regarding the upgrade: apt tells me: make is already the newest version (4.1-6).  Guess I have to upgrade Ubuntu first.
Will upgrade as soon as WSL-2 supports USB.

dpgeorge, 2019-12-18:
I'm happy to make it work with older make versions, as long as it also works with newer ones.

codefreax, 2020-01-01:
Ran into the same issue when compiling on the current OS X (10.15.2).

```
> make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
> which make
/usr/bin/make
```

UnexpectedMaker, 2020-01-01:
On on GNU Make 3.81 (On macOS) and don't have this error - I build MP from source all the time... IDF 3.x and 4.x - in fact, I built latest MP with IDF 4.x yesterday.

jedie, 2020-01-02:
I setup a docker image for compiling micropython here: https://github.com/jedie/docker-micropython

If any interests, the i can made a pull request to merge this into mainline.

dpgeorge, 2020-01-06:
Should be fixed by 99b8c1a93731771bd7d8801a8ea939922f3d2991, which assigns the result of the `$call` to a dummy variable.

closed 2020-01-06 by dpgeorge