micropython/micropython #5427: struct.pack doesn't check for value bounds

Bockwuhrst, 2019-12-16:
Setup: MicroPython v1.11-555-g80df377e9 on 2019-11-04; PYBv1.1 with STM32F405RG

```python
struct.pack('B', 256)  # returns b'\x00'
```

CPython errors out with `struct.error: ubyte format requires 0 <= number <= 255`

dpgeorge, 2019-12-17:
There are a few places in MicroPython where limits like this are not checked, but instead it does automatic truncation.  This is done to keep code size down, and also, partly, because on microcontrollers truncation of bit patterns is usually what you want.

Bockwuhrst, 2019-12-17:
I though this might be intentional but I didn't find it noticed in the [docs](http://docs.micropython.org/en/latest/genrst/modules.html#struct).

For library code where the implementation is hidden this behavior seems quite unfortunate though.
Either the library has to manually validate input (slow) or the library user might get unexpected truncation.

dpgeorge, 2019-12-19:
Related: #3204