micropython/micropython #5428: Why is only half of SPIRAM used?

gmos, 2019-12-16:
```
    switch (esp_spiram_get_chip_size()) {
        case ESP_SPIRAM_SIZE_16MBITS:
            mp_task_heap_size = 2 * 1024 * 1024;
            break;
        case ESP_SPIRAM_SIZE_32MBITS:
        case ESP_SPIRAM_SIZE_64MBITS:
            mp_task_heap_size = 4 * 1024 * 1024;
            break;
```
See the above snippet from ports/esp32/main.c  function mp_task.  If SPIRAM = 64MBit only half of the amount is used.  Is there a reason for this?  If so, a line of comment might be justified.

dpgeorge, 2019-12-17:
There's only enough address space to map up to 32Mbit.  64MBit is only supported using a special "himem API" and bank switching.

gmos, 2019-12-17:
Check.

closed 2019-12-17 by gmos