micropython/micropython #5432: Feature request: API call to change the REPL prompt from >>> to an arbitrary string

kentindell, 2019-12-17:
I have a setup with several PyBoards and the /dev/tty moves about depending on boot order and it's confusing which terminal goes to which board. It would be really nice to be able to set >>> to something else. Perhaps even "A:\>" and "C:\>" (OK, maybe not that).

peterhinch, 2019-12-17:
How about a file **me.py**, customised for each board, with a function that prints its name?

kentindell, 2019-12-17:
That's what I do at the moment, but it requires you have to type a command to ask. I was thinking of something that made it dead obvious. Instead of hardwiring '>>>' a modifiable string would even allow colours to be used in the prompt.

The Bourne shell has had PS1 since the '70s. I was just thinking of something similar for REPL.

mikewadsten, 2019-12-17:
CPython has this feature in `sys.ps1` and `sys.ps2`: https://docs.python.org/3/library/sys.html#sys.ps1

I could see this being an optional feature added to either `modsys.c` or `modmicropython.c`. I know that supporting `sys.ps1 = "C:> "` takes more code to implement than say `micropython.prompt("C:> ")` so the latter might be more suitable. And the call could be added to a device's `boot.py` or `main.py` file.

hardkrash, 2019-12-17:
As a workaround for your specific use case.

I use miniterm.py a cli tool from pyserial with hwgrep:// url for opening to specific serial ports.
It can be based on most of the USB descriptor space: e.g. location, serial number, and vid+pid specific USB location matching.

With my different boards I even match on the product string from USB.

mikewadsten, 2019-12-17:
Or, adding on @hardkrash's comment, another alternative would be to set up udev rules to add a symlink for each board.

With just one board plugged in:

```sh
$ udevadm info /dev/ttyACM0 # or whatever the comm port shows up as
# make note of the ID_SERIAL_SHORT value
```

```
# /etc/udev/rules.d/99-pyboard.rules

SUBSYSTEM!="tty", GOTO="pyb_end"

SUBSYSTEM=="tty", ENV{ID_SERIAL_SHORT}=="<value from above>", ENV{ID_VENDOR_ID}=="f055", ENV{ID_MODEL_ID}=="9800", SYMLINK+="pyboard1"
# repeat for each board

LABEL="pyb_end"
```

```sh
$ sudo udevadm control --reload
$ sudo udevadm trigger  # or just unplug and replug
```

dpgeorge, 2019-12-18:
> It would be really nice to be able to set >>> to something else.

I would find this feature quite useful.  Often working with multiple boards at the same time (eg one server, one client) it's hard to remember which prompt is which.

> CPython has this feature in `sys.ps1` and `sys.ps2`

If this feature were to be implemented, these variables would be the way to do it, to make it compatible with CPython.

For a long time there's been the desire to override `sys.stdin` etc (can't find the existing issue right now) so if that were implemented in a general way then adding `sys.ps1` and `sys.ps2` would be very easy.

closed 2026-05-23 by agatti