micropython/micropython #5434: gc_collect_root crash in VS2017 Release build

fei2020, 2019-12-18:
I use upy 1.11. Running of Debug build works well. But Release one(VS2017 Release build with /O2) crash at gc.c LN343
```
void gc_collect_root(void **ptrs, size_t len) {
    for (size_t i = 0; i < len; i++) {
    .......
    }
}
```

CUE: Parameter "size_t len" get a very big value.
gc.c LN320
```
void gc_collect_start(void) {
    .......

    // Trace root pointers.  This relies on the root pointers being organised
    // correctly in the mp_state_ctx structure.  We scan nlr_top, dict_locals,
    // dict_globals, then the root pointer section of mp_state_vm.
    void **ptrs = (void**)(void*)&mp_state_ctx;
    size_t root_start = offsetof(mp_state_ctx_t, thread.dict_locals);
    size_t root_end = offsetof(mp_state_ctx_t, vm.qstr_last_chunk);
    gc_collect_root(ptrs + root_start / sizeof(void*), (root_end - root_start) / sizeof(void*));
    .......
}
```
"(root_end - root_start) / sizeof(void*)" get a wrong value.

stinos, 2019-12-18:
Can you please provide more information: 32bit or 64bit build, is this a custom build or do you just build micropython.vcxproj without changes, do you have Python code to reproduce the issue?

fei2020, 2019-12-18:
1 32 build
2 yes. just build micropython.vcxproj without changes.
3 I just invoke mp_obj_new_float many times(60+FPS per second).

stinos, 2019-12-18:
>  I just invoke mp_obj_new_float in C++ many times with very high frequency

Just a note, you probably know, but the bare *frequency* at which you do this shouldn't matter since there should be only 1 thread calling the MicroPython API. Only the amount of times could matter. Or the relative frequency with respect to other function calls ors.

Anyway i cannot reproduce the issue. This Python code runs fine:
```
for i in range(1000000000):
  a = float(i)
```
and if I add this in main.c
```
for (size_t i = 0; i < 1000000000; ++i) {
    mp_obj_new_float(0.0);
}
```
it also runs fine.

The information in the issue you mentioned isn't clear enough,  i.e. on the forum dpgeorge mentioned already 1 problem with it, so now I don't know if you fixed that already, hence I don't know what you are using right now. So I'm really going to need the exact code you use to reproduce this, as well as the exact changes to the build.

> the process mem grow to 1.6+G.

Default MicroPython only allocates a fraction of that, and only once at the beginning, so the memory never grows. Which means the code you use is doing other things, possibly corrupting MicroPython's stack/memory area. Which could indeed lead to:

> some thing wrong at "(root_end - root_start) / sizeof(void*)"

do you mean that `root_end` became smaller than `root_start` ?