micropython/micropython #5440: Use mp_thread for IRQ simulation

nos86, 2019-12-19:
Hello,
I'm currently developing the hal for ESP32 CAN ( #5310 ). Unfortunately, according to my knowledge, the esp-idf framework has an internal managment of the interrupts without the possibility to add a callback to manage the isr at mp hal level. So, my idea is to create a task that monitors the event reporting variable (managed by esp-idf) in order to trigger a python callback.

Can you suggest me, any document, I can refer to develop this functionality?

Thank you in advance,
Salvo

dpgeorge, 2019-12-20:
> my idea is to create a task that monitors the event reporting variable (managed by esp-idf) in order to trigger a python callback.

Indeed the esp32 CAN driver doesn't look like it provides any callbacks, so this approach is needed.  One way to do it would be to create a small task (small stack) that waits forever on `can_receive()` and then when a message comes in notifies the main MicroPython task.  That would be more efficient than polling the event status, but requires a dedicated task per event.

> Can you suggest me, any document, I can refer to develop this functionality?

This scheme is already used by socket to create a callback (to Python) when a socket becomes readable, see `ports/esp32/modsocket.c`, grep for `MICROPY_PY_USOCKET_EVENTS`

nos86, 2019-12-20:
Polling of event status is necessary to provide some additional services like auto-restart (after bus-off) or provide information about errors (as py.CAN does).

I was looking at modsocket.c and I found how to manage the API for python callback, but nothing on how create a task in a micropythonic way.

jimmo, 2019-12-23:
> I was looking at modsocket.c and I found how to manage the API for python callback, but nothing on how create a task in a micropythonic way.

Have a look at esp32/network_ppp.c for an example of creating a worker task that waits for some low-level event and passes the data onwards. It also shows how to start/stop the task (via connect_py() and active()).

Note that if you want to invoke the MicroPython VM from your task, you'll need to do much more initialisation for the task (take a look at esp32/mpthreadport.c). But instead it would be better to just do the minimum possible and use mp_sched_schedule to make the main MicroPython task do the work. For an example of that, look at extmod/modbluetooth_nimble.c -- all of the callbacks there are executed from an IDF task, but they stash their data in the ringbuf then use mp_sched_schedule to invoke the actual Python callback.