micropython/micropython #5442: UART.readline() reads the buffer even when it does not end in a newline character

goatchurchprime, 2019-12-20:
The [UART documentation](https://docs.micropython.org/en/latest/library/machine.UART.html#machine.UART.readline) says:

```
UART.readline()
    Read a line, ending in a newline character.
    Return value: the line read or None on timeout.
```

If I connect a wire between TX2 and RX2 on the ESP32, and go:
```Python
from machine import UART
import time
u = UART(2)
s = u.write("begin1 end1\nbegin2 ");
# s = u.write("end2\nbegin3 ");
time.sleep(0.1)
print(u.any(), u.readline())
print(u.any(), u.readline())
print(u.any(), u.readline())
print(u.any(), u.readline())
```

The output comes out as:
```
19 b'begin1 end1\n'
7 b'begin2 '
None
None
```
You can uncomment the second u.write() to prove that the strings will be concatenated in the buffer prior to the readline().

(The timeout value in the constructor is not documented, but I've worked out that it is in milliseconds.)

With quite a bit of fiddling I can prove the same effect on the ESP8266 with the following code loaded into `main.py`:

```Python
from machine import UART
import time, os

u2 = UART(1, baudrate=115200)
print("startup")
u2.write("startup\n")
time.sleep(3)  # allow for interrupt
print("3seconds")
u2.write("3seconds\n")

os.dupterm(None, 1)
u = UART(0, baudrate=115200, rxbuf=240)

s = u.write("begin1 end1\nbegin2 ");
s = u.write("end2\nbegin3 ");
time.sleep(0.1)
for i in range(5):
    u2.write("%d %d %s\r\n" % (i, u.any(), str([u.readline()])))
```

Then you can monitor the output of the TX1 from the ESP8266 (which goes out of the GPIO2) by connecting it to RX2 of a spare ESP32 which is running the program:

Then on the ESP32 you run:
```Python
from machine import UART
u = UART(2, timeout=5000)
while True:
    print(u.readline())
```
(Don't forget to connect the RX to TX of the ESP32 and the VIN and G pins from the ESP32 to the ESP8266 to send it the power.)

The result is:
```
b'...\x02startup\n'
b"0 1 [b'3seconds\\r\\n']\r\n"
b"1 1 [b'begin1 end1\\n']\r\n"
b"2 1 [b'begin2 end2\\n']\r\n"
b"3 1 [b'begin3 ']\r\n"
b'4 0 [None]\r\n'
```

The fact that UART.any() can return 1 even when there are >1 characters in the buffer is at least documented.  However, I can't tell how to use select.poll() for a sophisticated way of querying the available characters.  

Normally I'm not daft enough to waste so much time trying to use UART0 on the ESP8266 (owing to the problem that it blocks the Repl), but I'm trying to do something with a [Wemos Arduino Megas](https://freematics.com/store/index.php?route=product/product&product_id=91) that -- with the right selection of dip-switches -- wires the atmega to the esp8266 on its USB0, so I have no choice.  

Now that I have identified this issue (which was very hard to see when you can't print debug messages on a repl blocked esp8266), I can make some code to concatenate the incomplete lines.  But it would be much more useful if this was done properly inside the UART buffer itself.

goatchurchprime, 2019-12-20:
I made a readline buffering generator like this:

```Python
def uartreadline(uart, buffersize=240):
    buf = memoryview(bytearray(buffersize))
    e = 0
    while True:
        notyielded = True
        s, e = e, e + (uart.readinto(buf[e:]) or 0)
        while s < e:
            s += 1
            if buf[s-1] == ord('\n'):
                yield bytes(buf[:s])
                buf[:e-s] = buf[s:e]
                s, e = 0, e-s
                notyielded = False
        if e == len(buf):
            yield b"BUFFEROVERFLOW\n"
            e = 0
        elif notyielded:
            yield None
```

Unfortunately I had other issues with the ESP8266's UART0 in that it has a very tiny buffer that it will happily write bytes at random onto -- even though I'd set `rxbuf=240` in the constructor (so that's another bug that's quite hard to prove because it comes out as silently corrupted data).  

I was only able to make this Wemos Arduino mega system (with the Arduino wired to the ESP8266 UART0) actually work by inserting a `delay(1)` in between every serial write character so that my Micropython code had time to pull the bytes out of it before the buffer over-wrote itself.

jimmo, 2019-12-23:
I do agree that the documentation is misleading. The "ending in a newline character" implies that the returned string will always end in a newline (but I suspect it's supposed to be just simply defining what a "line" is).

So the current behavior is that it will keep reading bytes from the UART (which may be buffered by either a hardware or software buffer or both), waiting up to the timeout for each byte (although not all ports support a timeout -- see #5335 for related discussion), and then returning when either:
 - A timeout occurs (return whatevers been read so far)
 - A newline is read (return whole line)

(Note that the number of available bytes at the start of the call to readline() isn't used, bytes may come in during the readline operation).

Here's an example which I think illustrates that this behavior matches CPython.  ref https://docs.python.org/3/library/io.html#io.IOBase.readline

```
$ mkfifo /tmp/fifo
$ python3
>>> f = open('/tmp/fifo', 'r')
# In a different shell: $ echo -n 'hi' > /tmp/fifo
>>> f.readline()
'hi'
# $ echo 'hello' > /tmp/fifo
>>> f.readline()
'hello\n'
>>> 
```

If you wanted to change the behavior to "only ever return complete lines or None to indicate that a line wasn't available", it's tricky because if you run out of buffered bytes (and if using a timeout, no more bytes arrive in that interval), then you can't really put the bytes back into the low-level buffer, you'd need a second place to stash them for future calls. I guess that's exactly what your `uartreadline` solution is doing. It works well as a generator!

On the other hand, I guess you could make the uart timeout really big (effectively infinite)? In which case readline() would just block until a full line is available.

I'll have a go at updating the documentation.

> Unfortunately I had other issues with the ESP8266's UART0 in that it has a very tiny buffer that it will happily write bytes at random onto -- even though I'd set `rxbuf=240` in the constructor (so that's another bug that's quite hard to prove because it comes out as silently corrupted data).

Any extra info to try and repro this would be really useful...

goatchurchprime, 2019-12-24:
If it's not going to predictably break the incoming stream into complete lines (which is essential for many parsing operations), I'd vote for getting rid of the UART.readline() function entirely.  

If we're going to have to pick up the pieces for its unreliability outside of that class, then we might as well do it all outside of that class.  There is no use-case for this function if it *sometimes* works depending on the exact timing of the bytes from the sending process. 

Readlines works on files because all the bytes are already there, so we don't get these random breaks.  

The sys.stdin object is of type io.FileIO.  When you call readline() on this it hangs until you complete the line and hit the return key.  The single-key extreme slowness of the human on a keyboard is predictable and accounted for in the way it behaves.  

There is a strange function in the socket object called makefile() which returns a file object, but there is never any buffering in Micropython -- [according to the documentation](https://docs.micropython.org/en/latest/library/usocket.html#usocket.socket.makefile).  

Except that there is buffering when I code it myself (when I require it). 

I would prefer the Java solution where you wrap your IO stream in a completely separate [BufferedReader](https://stackoverflow.com/a/25270210) object, so that everyone who needs buffering can have buffering without having to write it for themselves.  

This would be a thing like that interesting wrapper class `memoryview(bytearray)` which nails it in terms of the right level of pragmatism.

(I'll try and recreate my UART0 lack of buffer size on the ESP8266 when I get home next week.)

peterhinch, 2019-12-25:
The **uasyncio** **StreamWriter** and **StreamReader** objects work well with UARTS and offer precise control over timeout behaviour. The coroutine reading the UART can block forever pending a line. Alternatively it can be subject to a timeout or cancellation as per the requirements of the application.

jason-neal, 2020-01-11:
#5491 updated the documentation to highlight readline may return on a timeout (configurable and in ms).