micropython/micropython #5443: sf2,sf6 & pyboard Build failure on latest 42e45bd

pacmac, 2019-12-21:
I have not changed my setup for over a month, and just updated, when I attempt to compile I get:

```
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: build-PYBV11/main.o: in function `stm32_main':
main.c:(.text.stm32_main+0x2e4): undefined reference to `f_mount'
arm-none-eabi-ld: build-PYBV11/factoryreset.o: in function `factory_reset_make_files':
factoryreset.c:(.text.factory_reset_make_files+0x1c): undefined reference to `f_open'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_make_files+0x2c): undefined reference to `f_write'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_make_files+0x32): undefined reference to `f_close'
arm-none-eabi-ld: build-PYBV11/factoryreset.o: in function `factory_reset_create_filesystem':
factoryreset.c:(.text.factory_reset_create_filesystem+0x28): undefined reference to `f_mkfs'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_create_filesystem+0x46): undefined reference to `f_setlabel'
arm-none-eabi-ld: build-PYBV11/storage.o: in function `pyb_flash_init_vfs':
storage.c:(.text.pyb_flash_init_vfs+0x34): undefined reference to `mp_fat_vfs_type'
arm-none-eabi-ld: build-PYBV11/sdcard.o: in function `sdcard_init_vfs':
sdcard.c:(.text.sdcard_init_vfs+0x44): undefined reference to `mp_fat_vfs_type'
make: *** [Makefile:573: build-PYBV11/firmware.elf] Error 1
```

This fails on Pyboard, SF2 SF6, however esp8266, esp32 and nrf are OK, so it is appears to be pyboard specific.

So it looks like something else has changed, any suggestions ??

pacmac, 2019-12-22:
Seems that it is caused by this: 799b6d1e0c5bfb2392b7978f549ab2c7d2e0cc29

Adding `MICROPY_VFS_FAT=1` to the make command seems to fix it !

peterhinch, 2019-12-22:
I created a new build for SF2 yesterday and used litllefs without problems. However I did a fresh clone of MicroPython source. Perhaps this will provide a clue to what's going on?

pacmac, 2019-12-22:
Hi Peter;

Yes, I did a fresh clone too, but as above adding: `MICROPY_VFS_FAT=1` fixed the build for me .

I also stepped forward commit by commit to find the one that caused it.

pacmac, 2019-12-23:
Peter, what are your fresh clone steps ??

jimmo, 2019-12-23:
@pacmac Can you show your exact clean and make commands?

I can build SF6 and pybv11 no problems.

pacmac, 2019-12-23:
Hi Jimmo;

Sam bash scripts that I have used for the last year !!

Everything works fine until https://github.com/micropython/micropython/commit/799b6d1e0c5bfb2392b7978f549ab2c7d2e0cc29 and thereafter it fails with the errors provided.

I have also tried manually deleting the clone folder to be 100% sure it is removed, and have tested it numerous times.

I can step forwards commit by commit from https://github.com/micropython/micropython/commit/f2ecfe8b83b4efe24d721778e324a31237653627 onwards and as soon as it hits 799b6d it fails.

Then adding `MICROPY_VFS_FAT=1` the builds succeed and esp8266, esp32 and nrf do not have any problems !

```
function build {
  echo; echo "@@ Building micropython ..."

  ## FIX FOR ENV VARIABLE CORRUPTIONS (damien)
  cd "$UPYDIR/../../"
  make -C mpy-cross
  
  cd "$UPYDIR"
  
  MAKE="make -j8 FROZEN_MANIFEST=$PORTDIR/frozen.py $MAKE"
  
  if [ "$OPT" == "clean" ]; then
    echo "make clean..."
    $MAKE clean
  fi
  
  $BASHDIR/osxcln
  echo "$MAKE";
  $MAKE || exit 1
}
```

pacmac, 2019-12-23:
This is the make command with the translated variables:

`make -j8 FROZEN_MANIFEST=/nfs/qnap/dev/uPython/bash/../build/micropython/ports/stm32/frozen.py BOARD=PYBD_SF2 MICROPY_VFS_FAT=1`

pacmac, 2019-12-23:
Just tested again using manual commands:

**SUCCESS**
```
make clean 
make -j8 FROZEN_MANIFEST=/nfs/qnap/dev/uPython/bash/../build/micropython/ports/stm32/frozen.py MICROPY_VFS_FAT=1

LINK build-PYBV10/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 396396	     76	  55592	 452064	  6e5e0	build-PYBV10/firmware.elf
GEN build-PYBV10/firmware.dfu
GEN build-PYBV10/firmware.hex
```

**FAILURE**
```
make clean 
make -j8 FROZEN_MANIFEST=/nfs/qnap/dev/uPython/bash/../build/micropython/ports/stm32/frozen.py
LINK build-PYBV10/firmware.elf
arm-none-eabi-ld: build-PYBV10/main.o: in function `stm32_main':
main.c:(.text.stm32_main+0x2e4): undefined reference to `f_mount'
arm-none-eabi-ld: build-PYBV10/factoryreset.o: in function `factory_reset_make_files':
factoryreset.c:(.text.factory_reset_make_files+0x1c): undefined reference to `f_open'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_make_files+0x2c): undefined reference to `f_write'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_make_files+0x32): undefined reference to `f_close'
arm-none-eabi-ld: build-PYBV10/factoryreset.o: in function `factory_reset_create_filesystem':
factoryreset.c:(.text.factory_reset_create_filesystem+0x28): undefined reference to `f_mkfs'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_create_filesystem+0x46): undefined reference to `f_setlabel'
arm-none-eabi-ld: build-PYBV10/storage.o: in function `pyb_flash_init_vfs':
storage.c:(.text.pyb_flash_init_vfs+0x34): undefined reference to `mp_fat_vfs_type'
arm-none-eabi-ld: build-PYBV10/sdcard.o: in function `sdcard_init_vfs':
sdcard.c:(.text.sdcard_init_vfs+0x44): undefined reference to `mp_fat_vfs_type'
make: *** [Makefile:573: build-PYBV10/firmware.elf] Error 1
```

jimmo, 2019-12-23:
Hrmm... `MICROPY_VFS_FAT ?= 1` is set in `stm32/mpconfigport.mk`, and that's one of the first things included by `stm32/Makefile`.

Just to check, you don't have any additional `makefile` or `GNUmakefile` ?

And just to simplify as much as possible, what if you don't include the `FROZEN_MANIFEST`, i.e. just `make` followed by `make clean`.

pacmac, 2019-12-23:
Yes, I have a GNUmakefile:

```
include Makefile
CFLAGS += -Wno-error
```

pacmac, 2019-12-23:
make clean
make -j8

```
...
LINK build-PYBV10/firmware.elf
arm-none-eabi-ld: build-PYBV10/main.o: in function `stm32_main':
main.c:(.text.stm32_main+0x2e4): undefined reference to `f_mount'
arm-none-eabi-ld: build-PYBV10/factoryreset.o: in function `factory_reset_make_files':
factoryreset.c:(.text.factory_reset_make_files+0x1c): undefined reference to `f_open'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_make_files+0x2c): undefined reference to `f_write'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_make_files+0x32): undefined reference to `f_close'
arm-none-eabi-ld: build-PYBV10/factoryreset.o: in function `factory_reset_create_filesystem':
factoryreset.c:(.text.factory_reset_create_filesystem+0x28): undefined reference to `f_mkfs'
arm-none-eabi-ld: factoryreset.c:(.text.factory_reset_create_filesystem+0x46): undefined reference to `f_setlabel'
arm-none-eabi-ld: build-PYBV10/storage.o: in function `pyb_flash_init_vfs':
storage.c:(.text.pyb_flash_init_vfs+0x34): undefined reference to `mp_fat_vfs_type'
arm-none-eabi-ld: build-PYBV10/sdcard.o: in function `sdcard_init_vfs':
sdcard.c:(.text.sdcard_init_vfs+0x44): undefined reference to `mp_fat_vfs_type'
make: *** [Makefile:573: build-PYBV10/firmware.elf] Error 1
```

pacmac, 2019-12-23:
OOOh, I just found it, my build scripts over-write the mpconfigport.mk

```
# Enable/disable extra modules

## PAC 
MICROPY_PY_LWIP ?= 1

# wiznet5k module for ethernet support; valid values are:
#   0    : no Wiznet support
#   5200 : support for W5200 module
#   5500 : support for W5500 module
MICROPY_PY_WIZNET5K ?= 5500

# cc3k module for wifi support
MICROPY_PY_CC3K ?= 0
```

jimmo, 2019-12-23:
OK glad you found it!

FYI, you probably should be using the boards feature rather than needing to overwrite any of the files. i.e. you can make a custom board directory, that overrides both the config headers (mpconfigport.h) and makefiles (mpconfigboard.mk), but also extends from the included ones (e.g. PYBV11). Look at how the SF6 board extends the SF2 board for an example.

Note that a custom board definition can live in a totally different directory (i.e. in a separate git repo).

closed 2019-12-23 by jimmo

pacmac, 2019-12-23:
Thanks but I dont understand: 

> Look at how the SF6 board extends the SF2 board for an example.

jimmo, 2019-12-23:
> Thanks but I dont understand:
> 
> > Look at how the SF6 board extends the SF2 board for an example.

Many of the individual files in boards/PYBD_SF6 include (or import / reference) their counterparts from boards/PYBD_SF2.

pacmac, 2019-12-23:
But that would mean that I still have to write file(s) to the boards/PORT folder that would overwrite an existing file in that folder right ???

jimmo, 2019-12-23:
Let's say you have the MicroPython repo in `~/src/micropython` and your custom board definition in `~/src/projects/myboard` (with mpconfigboard.h, mpconfigboard.mk, pins.csv, stm32fyxx_hal_conf.h, etc).

Then
`make BOARD=myboard BOARD_DIR=~/src/projects/myboard`
will use your "out of tree" board config.

The example I was giving is that you can (if you want) include the makefile fragments or headers from other boards from your ones. Although it's perhaps easier just to copy them and modify as necessary. Should not require modifying anything in `~/src/micropython`.

peterhinch, 2019-12-23:
That is **very** useful information. It would be good if it were documented.

pacmac, 2019-12-23:
wow that is really great and will mean that I can scrap several of my bash build scripts.

But I have a question, lets say I create a override mpconfigport.mk file (or other file) whill the build process first read the default file and then override the default with any values I have set, or will it only read my file and ignore the default ?

pacmac, 2019-12-23:
do we need to open a thread on the forum to discuss this ??

I just added a GNUmakefile and mpconfigport.mk to my custom folder and then included that folder in the make command, and it failed:

```
make -j8 BOARD_DIR=/nfs/qnap/dev/uPython/bash/../build/micropython/ports/stm32/custom FROZEN_MANIFEST=/nfs/qnap/dev/uPython/bash/../build/micropython/ports/stm32/frozen.py BOARD=PYBD_SF2 MICROPY_VFS_FAT=1
make: GNUmakefile: No such file or directory
make: *** No rule to make target 'GNUmakefile'.  Stop.
```

jimmo, 2019-12-23:
> do we need to open a thread on the forum to discuss this ??

Yes that would be good. Thanks

pacmac, 2019-12-23:
https://forum.micropython.org/viewtopic.php?f=2&t=7433