micropython/micropython #5451: umqtt.simple's check_msg triggers OSError -1 with TLS servers

SylvainGarrigues, 2019-12-23:
Please consider this code:

```python
import utime
from umqtt.simple import MQTTClient

mqtt = MQTTClient("test_mqtt_client_id", "test.mosquitto.org",
                  port=8883, keepalive=30, ssl=True)

def on_topic_updated(topic, msg):
    print((topic, msg))

mqtt.set_callback(on_topic_updated)
mqtt.connect()
mqtt.subscribe(b"Topic/For/Sylvain")
while True:
    print("Checking msg...")
    mqtt.check_msg()
    utime.sleep(1)
```
It systematically gives the same output (and I can reproduce it with another broker e.g. AWS IoT) 
```
$ /usr/local/Cellar/micropython/1.11/bin/micropython  main.py
Checking msg...
Checking msg...
Traceback (most recent call last):
  File "main.py", line 15, in <module>
  File "/Users/sylvain/.micropython/lib/umqtt/simple.py", line 204, in check_msg
  File "/Users/sylvain/.micropython/lib/umqtt/simple.py", line 173, in wait_msg
OSError: -1
```
What's strange is that if I use port 1883 and ssl=False (i.e. no encryption), the same code works.

I reproduced this issue on my Mac (MicroPython 1.11) and on my ESP8266 board (MicroPython 1.12).

Looping over `mqtt.wait_msg()` works like a charm, so I suspect switching back and forth a TLS socket from blocking to non-blocking generates such error.

MrSurly, 2019-12-23:
Which board?

SylvainGarrigues, 2019-12-23:
I reproduced this issue on my Mac (MicroPython 1.11) and on my LoLin NodeMCU v3 board from wemos.cc (MicroPython 1.12).

peterhinch, 2019-12-24:
TLS on nonblocking sockets is a known issue, mentioned in my comment [here](https://github.com/micropython/micropython/issues/4821#issuecomment-535877938). My experience is that TLS on nonblocking sockets works on Pyboard D but not on ESPx.

SylvainGarrigues, 2019-12-24:
I thought 
https://github.com/micropython/micropython/commit/c76445315f836eb75835814ebe97e14107adbf0f was implementing non-blocking sockets for my board, doesn’t it?

MrSurly, 2019-12-24:
>TLS on nonblocking sockets is a known issue, mentioned in my comment here. My experience is that TLS on nonblocking sockets works on Pyboard D but not on ESPx

It's implemented on ESP32.

odewdney, 2020-02-15:
i was just about to create a pull request to update axtls, for this, when i found that its already been fixed in 
https://github.com/pfalcon/axtls/commit/f38bdcb22c23d2721adf0f6b659e57be21f13e93
(and other places!) - basically it was sending a tls alert on EAGAIN, killing the connection.

jordanarpad, 2022-09-03:
I see the same OSerror:-1 with umqtt.simple MQTTClient if I subscribe with the same client id that the message is published with

jonnor, 2024-09-02:
Already fixed, according to above comments.

closed 2024-09-02 by jonnor