micropython/micropython #5454: ESP32 4MB RAM Guru Meditation Crash On Initializing UART

Trekintosh, 2019-12-25:
Good Evening and Merry Christmas!

I was doing an ESP32 Micropython project that happens to use a serial port (UART 1) and decided I needed more RAM. After ordering a TTGO T8 V1.7 (ESP32, 4MB RAM, integrated MicroSD port) and setting up the most recent MicroPython build for SPI RAM, it would always reboot with a guru meditation error when I ran my code. After some poking around I found out that either declaring or initiating a UART would cause a guru meditation error and reset the ESP32. On the same board, the non-SPI RAM build does *not* do this. I tested this on the 3 most recent ESP32 SPIRAM builds from the download page.

Here's an example of this error:

```
MicroPython v1.12 on 2019-12-20; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from machine import UART
>>> uart = UART(1,baudrate=19200)
[0;32mI (27310) uart: ALREADY NULL[0m
>>> 
>>> 
>>> uart.init()
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dfc04  PS      : 0x00060930  A0      : 0x800d2d91  A1      : 0x3ffc2300  
A2      : 0x3f8177c0  A3      : 0x000000b1  A4      : 0x3ffc2320  A5      : 0xffddcccc  
A6      : 0x00000001  A7      : 0x000000ff  A8      : 0x00000019  A9      : 0x3ffc22e0  
A10     : 0xffddcccc  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffb32c0  
A14     : 0x00000003  A15     : 0x3ffc24c0  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffddcce8  LBEG    : 0x400d2fa8  LEND    : 0x400d3000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400dfc04:0x3ffc2300 0x400d2d8e:0x3ffc2320 0x400f5c30:0x3ffc2360 0x400d3003:0x3ffc2380 0x400d28dd:0x3ffc23a0 0x400d67a1:0x3ffc23c0 0x40104b32:0x3ffc2440 0x40104db4:0x3ffc24e0 0x400f5b1c:0x3ffc2520 0x40092d59:0x3ffc2550

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4844
load:0x40078000,len:10532
load:0x40080400,len:6736
entry 0x400806f0
[0;32mI (534) psram: This chip is ESP32-D0WD[0m
[0;32mI (534) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (534) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (537) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (544) cpu_start: Pro cpu up.[0m
[0;32mI (548) cpu_start: Application information:[0m
[0;32mI (553) cpu_start: Compile time:     Dec 20 2019 07:52:43[0m
[0;32mI (559) cpu_start: ELF file SHA256:  0000000000000000...[0m
[0;32mI (565) cpu_start: ESP-IDF:          v3.3[0m
[0;32mI (570) cpu_start: Starting app cpu, entry point is 0x40083d90[0m
[0;32mI (562) cpu_start: App cpu up.[0m
[0;32mI (1446) spiram: SPI SRAM memory test OK[0m
[0;32mI (1446) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1446) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1453) heap_init: At 3FFBA658 len 000259A8 (150 KiB): DRAM[0m
[0;32mI (1459) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1465) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1472) heap_init: At 400973F0 len 00008C10 (35 KiB): IRAM[0m
[0;32mI (1478) cpu_start: Pro cpu start user code[0m
[0;32mI (155) cpu_start: Chip Revision: 1[0m
[0;33mW (155) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.[0m
[0;32mI (158) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
MicroPython v1.12 on 2019-12-20; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> 
>>> 
```

Apologies for odd formatting, that was copied straight from upycraft and it handles characters a bit weirdly.

robert-hh, 2019-12-25:
The default pin assignments for UART 1 collide with the GPIO pins of the RAM. When instantiating UART, you have to specify other pins for tx and rx, like 4 and 5.

Trekintosh, 2019-12-25:
> The default pin assignments for UART 1 collide with the GPIO pins of the RAM. When instantiating UART, you have to specify other pins for tx and rx, like 4 and 5.

Thanks! That was the issue, and I feel like a doofus. In my project I changed tx and rx when I initialized the chatpad, but *not* when I instantiated it, which is when it was actually crashing for me. Working good now!

robert-hh, 2019-12-25:
Since this is a known trap, it should have been changed since long.

moefear85, 2021-06-07:
It's not a trap, atleast not as of 08.06.2021. You're flashing the wrong micropython image onto a board with spi-ram. Use the spiram image instead, then UART(1, 9600) works as-is.

moefear85, 2021-06-07:
That leaves the question, how to assign the pins before opening the UART? Calling the constructor without any arguments doesn't work, and it doesn't accept in arguments in the constructor, but that's where the exception happens, even before UART.init() is called