micropython/micropython #5457: Enabling Control-C in a new port

osresearch, 2019-12-25:
[image]

I've had success in [porting MicroPython to the EFR32 ](https://github.com/osresearch/micropython/tree/efm32/ports/efm32), although something that I'm stumbling on now is how to enable interrupting the execution with `^C` while a long running process is executing.

Running `import micropython ; micropython.kbd_intr(3)` doesn't seem to enable it, although it looks like the VM isn't checking the status of `mp_hal_stdio_poll()` during the loop, so nothing would notice that an interrupt character was sent.

Does this requiring enabling RX interrupts or is there a way to enable it without doing so?

dpgeorge, 2019-12-25:
> how to enable interrupting the execution with `^C` while a long running process is executing.

That's something which is not built-in to the core, rather each specific port must implement this feature.

> Does this requiring enabling RX interrupts or is there a way to enable it without doing so?

For efficiency it's best to do it using a hardware IRQ (eg UART RX interrupt).  For example:
- stm32 USB VCP: checks for `mp_interrupt_char` in `usbd_cdc_receive()` (on a hardware IRQ) and calls `mp_keyboard_interrupt()` if the char comes in.
- stm32 UART: same as above, on UART RX IRQ
- esp8266 UART: in `uart0_rx_intr_handler()` checks for `mp_interrupt_char`
- esp32 UART: in `uart_irq_handler()` checks for `mp_interrupt_char`
- webrepl: incoming TCP packet will call an asynchronous callback which ends up in `mp_uos_dupterm_rx_chr()` that checks for `mp_interrupt_char` and calls `mp_keyboard_interrupt()`

But it's also possible to add hooks in the port to poll for incoming chars and check for ctrl-C, eg in the javascript port, `MICROPY_VM_HOOK_POLL` calls `mp_js_hook()` which checks for ctrl-C (should also be checked in `MICROPY_EVENT_POLL_HOOK`).

If you can, I'd recommend to do it in the UART RX IRQ.

osresearch, 2020-01-12:
Thanks!  Enabling RX IRQ was easy enough for the EFM32 port: https://github.com/osresearch/micropython/commit/a7539534af19f427056d8558cef36ea2cd1c02a8

closed 2020-01-12 by osresearch