micropython/micropython #5467: pyboard-D AP_IF - Unable to set up a secure wlan

swehog, 2019-12-28:
Unable to setup a secure wlan (password protected), the password config param is there for this reason?

The code below are from the example at pybd.io.

MicroPython v1.12-23-ge83ab7374 on 2019-12-27; PYBD-SF2W with STM32F722IEK

`import network

wl_ap = network.WLAN(1)
wl_ap.config(essid='PYBD')                  # set AP SSID
wl_ap.config(password='pybd0123')   # set AP password
wl_ap.config(channel=6)                     # set AP channel
wl_ap.active(1)                                     # enable the AP`

dpgeorge, 2019-12-28:
In what way were you not able to set up a secure wlan AP?  What did not work and what did you expect to happen?

IveJ, 2019-12-28:
Hi Swehog,

Depend on your instance declare which interface type/mode, you can specific
auth for AP_IF.

swehog, 2019-12-28:
Hi,

The wlan AP will go active with the code above but as an open network without WPA. = Everything except the WPA-protection seems to work. (I expect the WLAN to have WPA protection.)

swehog, 2019-12-28:
> Hi Swehog, Depend on your instance declare which interface type/mode, you can specific auth for AP_IF.

I can not find out how to make this happen, i have tried with param "auth_mode" and "authmode" without success all i get is "ValueError: unknown config param"

IveJ, 2019-12-28:
Hi Seghog,

https://docs.micropython.org/en/latest/library/network.WLANWiPy.html

https://randomnerdtutorials.com/micropython-wi-fi-manager-esp32-esp8266/

Happy code.

swehog, 2019-12-28:
> Hi Seghog,
https://docs.micropython.org/en/latest/library/network.WLANWiPy.html https://randomnerdtutorials.com/micropython-wi-fi-manager-esp32-esp8266
/ Happy code.

Hi,

Neither of these examples does work on my pyboard-D.

I get "ValueError: unknown config param" in WLAN.config(xParam=y) when i try to pass in one of the following params; auth, authmode, auth_mode

IveJ, 2019-12-29:
Hi Swehog,

You should reference to libraries of your specified case for pyboard.

> Hi Seghog,
> https://docs.micropython.org/en/latest/library/network.WLANWiPy.html
> https://randomnerdtutorials.com/micropython-wi-fi-manager-esp32-esp8266
> / Happy code.
>
> Hi Swehog,

Neither of these examples does work on my pyboard-D.

swehog, 2019-12-29:
Hi again,

I'm using the network.WLAN class which came with the prebuilt firmware for "PYBD_SF2W" downloaded at https://micropython.org/download. I just uploaded "PYBD-SF2-20191220-v1.12" to the PYBD.

I am trying to get this to work with the "To use the WiFi access-point interface" reference code from http://pybd.io. It doesn't state that the password will make the wifi wpa protected but I can't see why we would need a password if it wouldn't. I can pass in essid, password, and channel, the accesspoint will go live but as an open network (no WPA). I can't pass in auth or authmode which are documented for other modules.

Pyboard D-series boards are using the CYW43 BT/WIFI module from murata, for which there are code available in this repo. I don't know how micropython are using those C files so i'm not qualified to investigate why this won't work.

swehog, 2019-12-31:
@dpgeorge @IveJ 
I did a build with the auth parts from pull request #5170 which fixed my problems.

`import network

ap = network.WLAN(1)
ap.active(0)
ap.config(essid='PYBD')
ap.config(password='pybd0123')
ap.config(channel=1)
ap.config(authmode=4)
ap.active(1)`

ammgws, 2021-07-27:
Is this an issue that only affects a small minority of boards that have different WIFI chips? My random NodeMCU clone board also experiences this issue but I was surprised to come across this and find that there is hardly any discussion.